ARMmbed/mbed-cli #607: improve test speed

davidhyman, 2018-01-11:
Local test runs currently take ~20 minutes - it would be easier to develop if they were faster.

theotherjimmy, 2018-01-11:
@davidhyman There is always a trade off between speed and completeness. You are welcome to suggest enhancements for the speed, as long as they maintain (or improve) the completeness of the tests.

screamerbg, 2018-03-23:
Some stuff was introduced to improve `py.test test` in Mbed CLI 1.5.0

screamerbg, 2018-04-28:
The py.test tests have been reduced to about 1:30 minutes with the Mbed CLI 1.5.0 release. Hope that solves the issue. The extended tests that include importing of remote repositories, compile etc, take another 10 minutes, but could be executed separately and primarily exercise other Mbed OS tools like the build system, exporters, supported targets etc.

closed 2018-04-28 by screamerbg