ARMmbed/mbed-cli #615: Should not do full rebuild when switching profiles

janjongboom, 2018-01-18:
Right now when you switch profiles it'll trigger a full rebuild. That is very annoying, I'm switching quite often between debug and release builds to look at memory usage etc. Can we not do this?

Downside is that changing the paths to `BUILD/PLATFORM/GCC_ARM/PROFILE/` requires change in documentation.

mark-edgeworth, 2018-01-18:
alternative might be to add '-profilename' to the end of either the platform or the compiler parts in the BUILD structure, although this feels a bit hacky.

geky, 2018-01-18:
You could create a symbolic link for the old path every build. Would be a cheap way to keep the old paths working for a while. The symbolic links could be removed at a later timer (6.0?)

janjongboom, 2018-01-19:
I don't know if Windows properly supports them.

screamerbg, 2018-01-29:
cc @theotherjimmy

theotherjimmy, 2018-01-29:
@janjongboom I don't think in re-builds assembly. I'll freely admit that is at best a 2% speedup. @geky is correct in pointing out that we would have to make the final artifact available as part of the `BUILD/<target>/<toolchain>/` directory. Maybe just copying them would be sufficient? I'm open to ideas on this one.

bmcdonnell-ionx, 2018-06-14:
> we would have to make the final artifact available as part of the `BUILD/<target>/<toolchain>/` directory.

I think you would want to do something like

`BUILD/<target>/<toolchain>/<build-profile-name-without-json-extension>/`

as I intended to suggest on #700.

bmcdonnell-ionx, 2018-06-14:
>> You could create a symbolic link...
>
> I don't know if Windows properly supports them.

Several (most?) versions of Windows require administrator rights to create symbolic links. I don't think you want to impose that requirement on your users. (Please don't.)

janjongboom, 2018-06-19:
@bmcdonnell-ionx Yes, exactly same proposal.

bmcdonnell-ionx, 2018-08-06:
#712 has been merged to `master`, though not released yet.

Should this issue be closed?

theotherjimmy, 2018-09-24:
Yes.

closed 2018-09-24 by theotherjimmy