ARMmbed/mbed-cli #617: Doxygen command

geky, 2018-01-22:
Is there a way to generate doxygen with mbed CLI?

Like the testing and config commands, a doxygen command would be quite helpful

cc @theotherjimmy, @dannybenor

theotherjimmy, 2018-01-22:
Nope. We could add that though.

screamerbg, 2018-05-26:
@sg- There are already doxyfiles inside Mbed OS that could be used as a default template for doxygen. What do you think?

theotherjimmy, 2018-05-29:
@screamerbg @geky  See the [`cargo doc` documentation](https://www.mankier.com/1/cargo-doc) for an instance of prior work. @screamerbg @sg- and I would have to work out what is appropriate behavoir for a `doc` subcommand.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-10-31 by ciarmcom