ARMmbed/mbed-cli #618: mbed export with "--source .." fails to discover mbed_app.json

Alex-EEE, 2018-01-24:
$ mbed export --source .. -i IAR
Scan: ..
Traceback (most recent call last):
...snip...
tools.config.ConfigException: Required parameter 'storage-selector.storage' defined by 'library:storage-selector' doesn't have a value

This happens despite having an mbed_app.json file that defines storage-selector.storage.

***
When I perform simply
mbed export -i IAR

The program finds the mbed_app.json file and finds storage-selector.storage

bridadan, 2018-01-25:
Hi Alex,

The default value for the `--source` option is just the current directory (`["."]`). If you supply a `--source` option, you are replacing this default with what you supply, in this case `[".."]`. If your `mbed_app.json` is located in your current directory (`"."`) instead of in your new source directory (`".."`), then the `mbed_app.json` will not be included. If you wish to use the `mbed_app.json` in its current location, I _believe_ you should be able to just point to it by adding the `--app-config <path to mbed_app.json>` to your `mbed export` command. Otherwise, you can just move your `mbed_app.json` into your new source directory.

Let me know how it goes and we can go from there, thanks!

Alex-EEE, 2018-01-29:
@bridadan thanks as always for a quick response!  Was actually working on that IP bridge we were discussing and had some interesting results!  Anyway, I'm back on this now, and yes, both of your suggested solutions ( move .json file to source target, or supply path with --app-config)  Note to anyone reading that the path to provided to app-config needs to be relative to where you invoked the CLI, not relative to where ever you targeted --source.  ex: $ mbed export --source .. -i IAR --app-config mbed_app.json

bridadan, 2018-01-30:
> Note to anyone reading that the path to provided to app-config needs to be relative to where you invoked the CLI, not relative to where ever you targeted --source. ex: $ mbed export --source .. -i IAR --app-config mbed_app.json

That's correct, thanks for underlining that fact! @Alex-EEE if that fixes your problem would you mind closing this issue?

closed 2018-01-30 by Alex-EEE