ARMmbed/mbed-cli #623: mbed publish not authenticating

mray19027, 2018-01-30:
Details:
When trying to execute `mbed publish` on an HG repository with no username/password set in ~/.hgrc, CLI fails with an abort stating:

```
$ mbed publish
abort: http authorization required for https://os.mbed.com/users/michaelray/code/XXX/
[mbed] Checking for local modifications...
[mbed] Pushing local repository "XXX" to remote "https://os.mbed.com/users/michaelray/code/XXX"
[mbed] ERROR: "hg" returned error code 255.
[mbed] ERROR: Command "hg push -q" in "C:\Users\micray01\OneDrive - ARM\Documents\temp\XXX"
```

Expected result:
Mbed CLI should bring a dialog box up or ask in command line for username and password to authenticate with which then gets passed to HG

screamerbg, 2018-02-06:
@mray19027 could you provide more information on this? E.g. Mercurial and Mbed CLI versions, environment, perhaps non-secret contents of .hgrc or .hgini

screamerbg, 2018-02-14:
@mray19027 ping

screamerbg, 2018-03-23:
Closing due to inactivity. Please reopen if needed

closed 2018-03-23 by screamerbg