ARMmbed/mbed-cli #624: Can't find any documentation for creating own libraries

loverdeg-ep, 2018-02-02:
I'm having trouble finding any documentation of the recommended procedure for creating libraries. with mbed-cli

theotherjimmy, 2018-02-02:
@loverdeg-ep Where did you first look for this information? (We'll probably put it there, so that you find it next time)

theotherjimmy, 2018-02-02:
@AnotherButler @screamerbg Do either of you know the answer to this question?

AnotherButler, 2018-02-02:
We have this content: https://os.mbed.com/docs/v5.7/tools/working-with-mbed-cli.html#compiling-static-libraries

loverdeg-ep, 2018-02-02:
@AnotherButler  we're actually just trying to figure out how to "libraryize" and existing git repository

There is an "mbed new --library" option we were evaluating but couldn't figure out how it was operating.

Eventually we think we've realized there isn't anything special about libraries and that we could just do "mbed add" on any old git repository to add it as a library.

However the existing documentation is worded such that we assumed an mbed library was some special thing that might have more meta-data than just what is found in the .lib file.

loverdeg-ep, 2018-02-14:
@screamerbg, @theotherjimmy, @AnotherButler 
Any more info on this?

We've weighed the pro's and con's and will probably be, reluctantly, switching back to using git sub-modules until @ilg-ul's [xPacks](https://xpack.github.io/l) or cmsis-packs become more production ready.

To continue my own education; Is there a technical write-up anywhere of how mbed-cli manages git repositories and "libraries"?

theotherjimmy, 2018-02-14:
@loverdeg-ep  would #626 Resolve the problem for you? If so, I recommend that you mention your support in that PR.  

-----

@loverdeg-ep I'm sure that @JonatanAntoni would like to hear what you think could help make cmsis-packs production ready for you at https://github.com/ARM-software/CMSIS_5

loverdeg-ep, 2018-02-14:
@theotherjimmy pulling in mbed-os as a submodule versus having mbed-cli do it is our next step so I'm all for it.

@JonatanAntoni eventually I may make an issue at https://github.com/ARM-software/CMSIS_5 but @ilg-ul but my thoughts on the topic aren't organized yet.

Overall @ilg-ul's vision for [xPacks](https://xpack.github.io/) is what we want period.

* package management tools
* build tool for building packs and projects using them
* A relatively easy to understand framework for describing and configuring packages

[Yotta](https://github.com/ARMmbed/yotta) also seemed like a good start, but I'm partial to json and javascript for all of this considering
* the success of NPM, node.js
* ease of jumping into javascript development from C/C++ (considering the similar syntax)

theotherjimmy, 2018-02-14:
@loverdeg-ep To be clear, I think that Keil uVision supports all three of your goals. Do you also want to use Mbed OS? That might be the rub then.

loverdeg-ep, 2018-02-14:
@theotherjimmy 
Yes for sure want to take advantage of hardware abstraction mbed OS provides.

Also not a fan of being locked into a particular IDE especially uVision.

We're currently waiting for the right moment to hop on the [eclipse Che](https://www.eclipse.org/che/docs/6/che/docs/) train since it has the potential to fulfill the grand vision of doing serious embedded development without installing a single tool on our individual workstations.

theotherjimmy, 2018-02-14:
Good to know. Thanks for all of the feedback.

theotherjimmy, 2018-09-24:
I think the question was answered. Please let us know if you have anything further.

closed 2018-09-24 by theotherjimmy