ARMmbed/mbed-cli #625: Calling mbed export with clean can cause crash

ssnover95, 2018-02-06:
I'm getting a crash when calling `mbed export` with the flag `--clean`. Here's the output:

```
ssnover@styrka:~/develop/mbed-os-example-blinky$ mbed export -i uvision --clean -m NUCLEO_F302R8
Traceback (most recent call last):
  File "/home/ssnover/develop/mbed-os-example-blinky/mbed-os/tools/project.py", line 259, in <module>
    main()
  File "/home/ssnover/develop/mbed-os-example-blinky/mbed-os/tools/project.py", line 249, in main
    rmtree(BUILD_DIR)
  File "/usr/lib/python2.7/shutil.py", line 239, in rmtree
    onerror(os.listdir, path, sys.exc_info())
  File "/usr/lib/python2.7/shutil.py", line 237, in rmtree
    names = os.listdir(path)
OSError: [Errno 2] No such file or directory: '/home/ssnover/develop/mbed-os-example-blinky/mbed-os/BUILD'
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/ssnover/develop/mbed-os-example-blinky/mbed-os/tools/project.py -i uvision -m NUCLEO_F302R8 -c --source ." in "/home/ssnover/develop/mbed-os-example-blinky"
```

On the line mentioned in `mbed-os/tools/project.py`, I noticed that the directory is never checked to be existing:

```
if options.clean:
        rmtree(BUILD_DIR)
```

There's no check that the BUILD_DIR exists. This mainly came up because I was passing `--source` flags and the setting of `BUILD_DIR` appears to rely on the default behavior of `--source .` being passed when the user doesn't use the flag. I was able to cause it in a newly created `mbed-os-example-blinky` by calling an export without ever calling compile (so the BUILD_DIR is never made).

screamerbg, 2018-02-06:
cc @theotherjimmy

theotherjimmy, 2018-02-13:
Hmmm. That's definitely a bug. At worst, it should tell you not to use `-c` or something.

theotherjimmy, 2018-02-13:
I'll have time later this week to look into this.

closed 2018-02-26 by cmonr