ARMmbed/mbed-cli #628: Cache does not work

janjongboom, 2018-02-19:
I have cache enabled:

```
$ mbed config -G cache
[mbed] on
```

However when I clone a project it still downloads Mbed OS.

```
janjon01:~/repos $ mbed import https://os.mbed.com/teams/mbed-x/code/SimpleBLE-Example-mbedos5/
[mbed] Importing program "SimpleBLE-Example-mbedos5" from "https://os.mbed.com/teams/mbed-x/code/SimpleBLE-Example-mbedos5" at latest revision in the current branch
[mbed] Adding library "SimpleBLE" from "https://developer.mbed.org/teams/mbed-x/code/SimpleBLE-mbed-os-5" at rev #450b780bb1de
[mbed] Adding library "mbed-os" from "ssh://git@github.com/armmbed/mbed-os.git" at rev #4c256f045961
Downloading |#----------------------------------------------------------| 2.87%
```

It should not download anything. I have cache enabled.

screamerbg, 2018-03-23:
@janjongboom Mbed CLI 1.5.0 is now release. Could you please install and try to reproduce?

screamerbg, 2018-05-17:
Bump @janjongboom