ARMmbed/mbed-cli #629: mbed add behaves different when changing to verbose mode (Win10)

geoffrey-vl, 2018-02-22:
I found that the **mbed add** command behaves different when changing to verbose mode, on Windows 10.
In non-verbose mode it asks the credentials of the remote user, and although provided correctly authentication fails.
In verbose mode it found my id_rsa, and authenticates.

See below:

```
C:\work-mbed\blinky>mbed add --depth 1 ssh://someserver/mbed-os.git mbed-os
[mbed] Adding library "mbed-os" from "ssh://someserver/mbed-os.git" at latest revision in the current branch
git@someserver's password:
[mbed] ERROR: Unable to clone repository (ssh://someserver/mbed-os.git/)
---

C:\work-mbed\blinky>mbed add --depth 1 -v ssh://someserver/mbed-os.git mbed-os
[mbed] Working path "C:\work-mbed\blinky" (program)
[mbed] Adding library "mbed-os" from "ssh://someserver/mbed-os.git" at latest revision in the current branch
[mbed] Exec "git clone ssh://someserver/mbed-os.git C:\work-mbed\blinky\mbed-os --depth 1" in C:\work-mbed\blinky
Cloning into 'C:\work-mbed\blinky\mbed-os'...
Enter passphrase for key 'P:\.ssh\id_rsa':
```

Note that my Windows machine has different id_rsa files. When using ssh it uses the C:/Users/me/.ssh folder. When cloning through git, it uses the P:/.ssh folder.

theotherjimmy, 2018-02-23:
@screamerbg Based on the description (different behavior based on -v), this looks very much like a bug to me.

@geoffrey-vl Is this blocking you in some way?

geoffrey-vl, 2018-02-23:
No, as work around I'm using verbose mode.
Could it have something to do with following function in mbed.py?

```
    def clone(url, name=None, depth=None, protocol=None):
        if verbose or very_verbose:
            popen([git_cmd, 'clone', formaturl(url, protocol), name] + (['--depth', depth] if depth else []) + (['-v'] if very_verbose else ([] if verbose else ['-q'])))
        else:
            pquery([git_cmd, 'clone', '--progress', formaturl(url, protocol), name] + (['--depth', depth] if depth else []), output_callback=Git.action_progress)
hide_progress()
```

screamerbg, 2018-02-23:
@geoffrey-vl Thanks for the bug report. Yes, it has to do with the progress bars formatting. I'll look into this tomorrow.

screamerbg, 2018-02-23:
@geoffrey-vl Hmmm, I'm not able to reproduce it. Here's what I use as a command:
```
D:\Temp>mbed import ssh://git@github.com/ARMmbed/mbed-os-example-client --protocol ssh
[mbed] Importing program "mbed-os-example-client" from "ssh://git@github.com/ARMmbed/mbed-os-example-client.git" at latest revision in the current branch
Enter passphrase for key '/.ssh/id_rsa':
[mbed] Adding library "easy-connect" from "ssh://git@github.com/ARMmbed/easy-connect.git" at rev #06594ba91bc1
Enter passphrase for key '/.ssh/id_rsa':
[mbed] Adding library "easy-connect\atmel-rf-driver" from "ssh://git@github.com/ARMmbed/atmel-rf-driver.git" at rev #ca9782e68f5f
Enter passphrase for key '/.ssh/id_rsa':
Enter passphrase for key '/.ssh/id_rsa':
[mbed] Adding library "easy-connect\esp8266-driver" from "ssh://git@github.com/ARMmbed/esp8266-driver.git" at rev #b0d79dad507d
Enter passphrase for key '/.ssh/id_rsa':
[mbed] Adding library "easy-connect\mcr20a-rf-driver" from "ssh://git@github.com/ARMmbed/mcr20a-rf-driver.git" at rev #93661a696735
```

Every prompt appears just fine. What git version do you use?

geoffrey-vl, 2018-02-23:
Well, it's on my desktop pc at work, can't tell right now but it should be a recent one.
What I do now is that our IT infrastructure has been setup so that the Windows "home" folder was actually a network folder mapped to the P drive, but this is no longer the case (at least not on my machine). The P drive is still around though, it is still mapped as P, I can still access it and some tools still use it as default home folder so it seems. As I said: when using ssh it uses the C:/Users/me/.ssh folder. When cloning through git, it uses the P:/.ssh folder. But I wouldn't expect mbed to have the same odd behavior when switching verbose mode on/off. Assuming that this is a Git issue, can you tell me the difference in git commands that are produced when adding libs through normal mode and through verbose mode? Maybe I can try and issue these directly, see if we get any wiser out of this.

geoffrey-vl, 2018-03-02:
I've asked our IT departed to completely remove the P: drive mapping and point everything to C:/Users/me. This seems to be okay, whe opening a promt I'm now taken to the home folder instead. However I'm running into issues. When I create a SSH key via Git-bash, and then run `mbed import`, I still get the same behavior as before: 

- **non verbose mode**: uses system's default SSH executable
```
C:\Users\me>ssh -V
OpenSSH_7.5p1, without OpenSSL

C:\Users\me>ssh someserver
key_load_public: invalid format
Load key "C:\\Users\\me/.ssh/id_rsa": invalid format
git@someserver's password:
```

- **verbose mode**: I guess this one goes through Git its SSH executable under C:\Program Files\Git\usr\bin
```
C:\Program Files\Git\usr\bin>ssh.exe -V
OpenSSH_7.5p1, OpenSSL 1.0.2k  26 Jan 2017

C:\Program Files\Git\usr\bin>ssh.exe someserver
Enter passphrase for key '/c/Users/me/.ssh/id_rsa':
Welcome to Ubuntu 16.04.3 LTS (GNU/Linux 4.4.0-101-generic x86_64)
Last login: Fri Mar  2 11:04:14 2018 from 10.5.3.50
```
Unfortunately as you may have notice in the output, the public key generated from one tool is not compatible with the other...

So in the end I'm still not sure why mbed-cli would call 2 different tools when issuing the same command.

theotherjimmy, 2018-03-02:
Huh, That is strange! Thanks for exploring this more deeply. @screamerbg You may have to install 2 incompatible ssh executables to reproduce this behavior.

geoffrey-vl, 2018-03-02:
What I've done is enable the SSH client that's available in Win10 since december last year. Some info on how it can be installed: https://fossbytes.com/enable-built-windows-10-openssh-client/

screamerbg, 2018-03-23:
@geoffrey-vl Mbed CLI doesn't invoke SSH at all, rather Git does. The main difference between the 2 commands is the `--progress` and `-v` flags. Could you please try plain `git clone <repo>` and `git clone <repo> --progress -v` and let me know?

geoffrey-vl, 2018-03-23:
Results:
git clone <repo> => Asks my SSH key
git clone --progress <repo> => Asks my SSH key
git clone --progress -v <repo> => Asks my SSH key
mbed import -v <repo> => Asks my SSH key
mbed import <repo> => Asks credentials

screamerbg, 2018-04-27:
@geoffrey-vl I'd like to revive this. I'm still not able to reproduce it on my local setup.

Verbose mode:
```
            popen([git_cmd, 'clone', formaturl(url, protocol), name] + (['--depth', depth] if depth else []) + (['-v'] if very_verbose else ([] if verbose else ['-q'])))
```

Non-verbose mode:
```
            pquery([git_cmd, 'clone', '--progress', formaturl(url, protocol), name] + (['--depth', depth] if depth else []), output_callback=Git.action_progress)
hide_progress()
```

There are 2 differences:
* `--progress` is used in non-verbose mode, e.g. when you're wrongly asked about credentials
* the process output is piped (`popen()` vs `pquery()`). Neither of these have anything to do with different git being executed and so on.

Based on your latest comment, `--progress` isn't the one to blame. Do you think you could pipe the process to a file or something, e.g. `git clone --progress https://repo >> test.log` or `git clone --progress https://repo | more` and see whether it behaves differently?

geoffrey-vl, 2018-06-08:
>  Do you think you could pipe the process to a file or something, e.g. git clone --progress https://repo >> test.log or git clone --progress https://repo | more and see whether it behaves differently?

>
Sorry for the delay, but I tested both commands and both work.

Also I repeated my previous test with a newer version of mbed-cli (1.5.1), but the outcome is that it isn't working either. The output:

**mbed import :**
`C:\Users\me\Downloads>mbed import ssh://someserver/mbed-os-example-filesystem
[mbed] Importing program "mbed-os-example-filesystem" from "ssh://someserver/mbed-os-example-filesystem" at latest revision in the current branch
Enter passphrase for key 'C:\Users\me/.ssh/id_rsa':
[mbed] ERROR: Unable to clone repository (ssh://someserver/mbed-os-example-filesystem)`

**mbed import -v :**
`C:\Users\me\Downloads>mbed import -v ssh://someserver/mbed-os-example-filesystem
[mbed] Working path "C:\Users\me\Downloads" (directory)
[mbed] Importing program "mbed-os-example-filesystem" from "ssh://someserver/mbed-os-example-filesystem" at latest revision in the current branch
[mbed] Exec "git clone ssh://someserver/mbed-os-example-filesystem C:\Users\me\Downloads\mbed-os-example-filesystem" in C:\Users\me\Downloads
Cloning into 'C:\Users\me\Downloads\mbed-os-example-filesystem'...
Enter passphrase for key '/c/Users/me/.ssh/id_rsa':
remote: Counting objects: 353, done.
remote: Compressing objects: 100% (217/217), done.
remote: Total 353 (delta 144), reused 322 (delta 129)
Receiving objects: 100% (353/353), 241.84 KiB | 10.99 MiB/s, done.
Resolving deltas: 100% (144/144), done.
...` 

Notice the difference when it's asking for the passphrase: **C:\Users\me/.ssh/id_rsa** vs **/c/Users/me/.ssh/id_rsa**

screamerbg, 2018-06-18:
@geoffrey-vl Thanks for the extended debug report. I'm certain that there's something wrong with Git, but I'm out of ideas how to reproduce it to find workaround for your specific issue. What version of Git do you use? And could you update to the latest one?

geoffrey-vl, 2018-06-21:
As a quick resume he're my current versions.
- via **cmd.exe**:
```
Microsoft Windows [Version 10.0.17134.112]
(c) 2018 Microsoft Corporation. Alle rechten voorbehouden.

C:\Users\me>ssh -V
OpenSSH_for_Windows_7.6p1, LibreSSL 2.6.4
C:\Users\me>git --version
git version 2.14.1.windows.1
```
- via **git-bash.exe**:
```
me@computer  MINGW64 ~
$ ssh -V
OpenSSH_7.5p1, OpenSSL 1.0.2k  26 Jan 2017

me@computer MINGW64 ~
$ git --version
git version 2.14.1.windows.1
```

And here is after installing latest git.
Note when installing I've used the "Use OpenSSH" option which make sure that git uses the ssh.exe as delivered with it (and will not use the system's openssh).
- via **cmd.exe**:
```
Microsoft Windows [Version 10.0.17134.112]
(c) 2018 Microsoft Corporation. Alle rechten voorbehouden.

C:\Users\me>ssh -V
OpenSSH_for_Windows_7.6p1, LibreSSL 2.6.4
C:\Users\me>git --version
git version 2.17.1.windows.2
```
- via **git-bash.exe**:
```
me@computer  MINGW64 ~
$ ssh -V
OpenSSH_7.5p1, OpenSSL 1.0.2k  26 Jan 2017

me@computer MINGW64 ~
$ git --version
git version 2.17.1.windows.2
```

Also note that in the mean time I've added another user on our internal git server so that it matches my Windows username, and I've removed shh config that made the username matching work before.

At this stage an mbed import seems to be **OK**:
```
C:\Users\me\Downloads>mbed import ssh://someserver/mbed-os-example-filesystem
[mbed] Importing program "mbed-os-example-filesystem" from "ssh://someserver/mbed-os-example-filesystem" at latest revision in the current branch
[mbed] Adding library "dataflash-driver" from "https://github.com/ARMmbed/dataflash-driver" at rev #efcb8a1a5c11
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #569159b784f7
[mbed] Adding library "sd-driver" from "https://github.com/ARMmbed/sd-driver" at rev #c16fa2bf36b8
[mbed] Adding library "spif-driver" from "https://github.com/ARMmbed/spif-driver" at rev #e5e6616914a2
```

I still need to do some extra tests to see if all is ok right now, keep you updated...

geoffrey-vl, 2018-06-21:
At this stage I think all is ok, I still have some other issue but I believe it should be handled in a different issue ticket.
Thanks

screamerbg, 2018-06-22:
@geoffrey-vl Great, thanks!

closed 2018-06-22 by screamerbg