ARMmbed/mbed-cli #630: parser for MBED_MEM_TRACE output

thinkberg, 2018-02-23:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

The current memtrace is a very nice feature, but rather hard to use. Since my forum post is now in moderation for a week, maybe you'd like to integrate my little script into the cli tooling itself?

https://github.com/thinkberg/mbed-memtrace-logger

It parses the MEM_TRACE output and displays allocated heap and +/- values in a more readable form to aid debugging, while keeping log functionality.

Leo.

theotherjimmy, 2018-02-23:
@thinkberg You're welcome to submit a PR. I'm not sure mbed CLI would be the best place for this. @screamerbg Do you have any thoughts on that subject?

thinkberg, 2018-02-23:
I remember something like `yotta debug` in the old system which could be helpful here too. An mbed client way to start openocd/jlink gdb server and maybe even log serial output in a file/pipe it through the the script I wrote if tracing is enabled. 

I didn't know which is the best part to submit something too, so I wanted to get some feedback. It felt more home in mbed-cli than in the mbed-os part.

screamerbg, 2018-02-23:
@thinkberg I agree with @theotherjimmy that Mbed CLI might not be the best place for the parsing. Do you think that this could be something like an external tool or python module, something that Mbed CLI could call? Or perhaps be part of Mbed OS tools given that many tools inside the OS have knowledge about the target, ram/flash, regions, even flash algos, device IDs, etc.

thinkberg, 2018-02-23:
Sure, could be an external module that can be called. My reasoning was that its important enough to be easily accessible. Right now, having a memory leak involves quite some searching and guessing. Even the stack/thread debug that I referenced in the repo was a lucky find. You almost always find outdates mbed docs, unfortunately. Even if it would be part of the docs with reference to the external tools would be most helpful.

screamerbg, 2018-05-26:
@thinkberg @theotherjimmy Mbed CLI now has a serial terminal feature (#664). Perhaps this is something that could be added to the serial terminal hooks and enable runtime memtrace functionality? Note that Mbed CLI does not have any external python dependencies, so the pygtail dependency should be somehow workarounded.

thinkberg, 2018-05-26:
Actually, pygtail was just a workaround due to buffering issues. If the mbed cli serial can pipe its data into a stream it would be simple to make it work.

thinkberg, 2018-05-26:
If you give me a hint where to integrate (where are the serial terminal hooks?), I can port it to be part of the cli.

theotherjimmy, 2018-05-29:
@thinkberg The file you're looking for is: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py

Let us know if you have any questions.

thinkberg, 2018-05-29:
I will take a look, thanks.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

thinkberg, 2018-11-14:
PR in #789

closed 2019-10-31 by ciarmcom