ARMmbed/mbed-cli #632: CMake export GCC_ARM .map

vidavidorra, 2018-02-27:
First of all, I'm not sure whether I should post this question here or in the mbed-os repository . I'm usign CLion as IDE and use the CMake exporter to generate the CMakeList files using
```shell
$ mbed export -i cmake_gcc_arm -m NUCLEO_F103RB
```
When I compile the code it generates the .hex file as expected, but I'm interested in the memory map of the code. To create that, I believe I need a `.map` file and use the [memap tool](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/concepts/memap/) to create output like below.

```
+------------------+-------+-------+-------+
| Module           | .text | .data |  .bss |
+------------------+-------+-------+-------+
| [fill]           |    62 |     4 |  2513 |
| [lib]/libc.a     | 22434 |  2204 |    56 |
| [lib]/libgcc.a   |  3728 |     0 |     0 |
| [lib]/libm.a     |    88 |     0 |     0 |
| [lib]/libnosys.a |    32 |     0 |     0 |
| [lib]/misc       |   232 |     8 |    28 |
| main.o           |    56 |     0 |     4 |
| mbed-os/features |    42 |     0 |   184 |
| mbed-os/hal      |   450 |     0 |     8 |
| mbed-os/platform |  1252 |     4 |   269 |
| mbed-os/rtos     |  5955 |    24 |  6874 |
| mbed-os/targets  |  9788 |    12 |   384 |
| Subtotals        | 44119 |  2256 | 10320 |
+------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 12576 bytes
Total Flash memory (text + data): 46375 bytes
```

Am I correct to say that the CMake exporter doesn't create a .map file and memory memory usage as shown above. It it possible to generate such output (a `.map` file) by default using the CMake exporter and how would I do so?

Mbed CLI: 1.2.2
Mbed OS: 5.7.4
Compiler: GNU ARM embedded (6-2017-q1-update 6.3.1 20170215)

theotherjimmy, 2018-02-27:
@vidavidorra I don't think that exporter creates a map file. You could add the linker flag `"-Wl,-Map=%s"` replacing `%s` with the name of your map file. That would generate the map file for you.

vidavidorra, 2018-02-28:
@theotherjimmy Thanks for your response. I've added the linker flag by adding the following line to my `project.cmake` and that successfully generates a `mbed-os-example-blinky.map` file.

```cmake
SET(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -Wl,-Map=${CMAKE_PROJECT_NAME}.map")
```

When I run the memap tool from `mbed-os/tools` I get the following output. There are a lot of unknown objects found.  I've tried both running it using python2 and 3 and tried running it from the root directory and `cmake-build-debug` directory, but all with the same result. See below for my folder structure.

```shell
$ python2 ../mbed-os/tools/memap.py mbed-os-example-blinky.map -t GCC_ARM
Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_STM/TARGET_STM32F2/TARGET_PARTICLE_ELECTRON/device/TOOLCHAIN_GCC_ARM/startup_stm32f207xx.S.obj
Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/main.cpp.obj
Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/main.cpp.obj

...

Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_STM/TARGET_STM32F2/serial_device.c.obj
Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_STM/serial_api.c.obj
Unknown object name found in GCC map file: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_STM/us_ticker_32b.c.obj
+-------------+-------+-------+------+
| Module      | .text | .data | .bss |
+-------------+-------+-------+------+
| [fill]      |    64 |     4 |   14 |
| [lib]/c.a   | 24933 |  2472 |   89 |
| [lib]/gcc.a |  3868 |     0 |    0 |
| [lib]/misc  |   252 |    16 |   28 |
| [misc]      | 32605 |   188 | 7001 |
| Subtotals   | 61722 |  2680 | 7132 |
+-------------+-------+-------+------+
Total Static RAM memory (data + bss): 9812 bytes
Total Flash memory (text + data): 64402 bytes
```

```
.
├── BUILD
├── CMakeLists.txt
├── cmake-build-debug
│   └── CMakeFiles
├── main.cpp
├── mbed-memory-status
│   └── RTT
└── mbed-os
    ├── TESTS
    ├── cmsis
    ├── drivers
    ├── events
    ├── features
    ├── hal
    ├── platform
    ├── rtos
    ├── targets
    └── tools
```

theotherjimmy, 2018-02-28:
@vidavidorra That's probably because memap.py assumes that object willl be named `*.o`. It looks like yours are named `*.obj`.

vidavidorra, 2018-02-28:
@theotherjimmy Can I do something about the name extension. I haven't changed anything from the default export as described in my initial post.
Manually changing `.c.obj`, `.cpp.obj` and `.S.obj` to `.o` seems to solve the problem and generates the following output (without the unknown object errors).

```
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |    72 |     4 |   10 |
| [lib]/c.a        | 24817 |  2472 |   89 |
| [lib]/gcc.a      |  3864 |     0 |    0 |
| [lib]/misc       |   208 |    12 |   28 |
| main.o           |   328 |     4 |   28 |
| mbed-os/drivers  |   536 |     4 |  100 |
| mbed-os/hal      |  2776 |     4 |   66 |
| mbed-os/platform |  2989 |     4 |  270 |
| mbed-os/rtos     | 15215 |   168 | 5989 |
| mbed-os/targets  | 10237 |     4 |  360 |
| Subtotals        | 61042 |  2676 | 6940 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 9616 bytes
Total Flash memory (text + data): 63718 bytes
```

Would changing the [regular expression](https://github.com/ARMmbed/mbed-os/blob/master/tools/memap.py#L81-L82) in the `memap.py` tool in general would be acceptable for this? I think changing it from `r'^(.+\/.+\.o)$'` to `r'^(.+\/.+\.o\(bj\)?\)$'` should be enough.
If this is an acceptable change I'll be happy to make the changes and create a pull request for this.

theotherjimmy, 2018-02-28:
Sure, that sounds good.

vidavidorra, 2018-03-01:
@theotherjimmy I've made the changes and it works great. This should actually be a pull request in [ARMmbed/mbed-os](https://github.com/ARMmbed/mbed-os) as that is where the `memap.py` tool is, do you want me to make a ticket there referring to this ticket?
From which branch do you want me to make the changes and to which branch do you want me to create the pull request?

theotherjimmy, 2018-03-01:
@vidavidorra No need to make a new ticket; this one will do fine. From a branch on your fork of mbed-os to mbed-os master.

vidavidorra, 2018-03-01:
@theotherjimmy I've made the changes and am ready to make a pull request. The branch I want to mere is [this](https://github.com/withthegrid/mbed-os/tree/feat/cmake_memap), but when I try to create a pull request to ARMmbed/mbed-os/master it shows 400 commits and 1200+ files changed. So I'm not sure whether that's good. The only file I've changed is `memap.py`.
How should I make the pull-request?

theotherjimmy, 2018-03-02:
@vidavidorra It looks like your branch is based on a release. Our releases cherry pick bug fixes from master. If you submit a PR against master from a branch based on a release branch, your PR will include the **entire release history** because the commits, while they contain the same contents, have different time stamps and parent commits which causes them to have a different hash.

-----

You can resolve this by cherry-picking your commit onto a fresh checkout of master.

Supposing:
 * the armmbed/mbed-os git repo is on the remote "armmbed"
 * The only commit you care about is the commit at HEAD of `feat/cmake_memap` (It looks like it)
 * You will only use that branch for the PR

Steps:
 1. `git checkout armmbed/master` Move your HEAD to upstream's master
 2. `git cherry-pick feat/cmake_memap` Create a new commit (no branch yet) from the diff on your feature branch
 3. `git branch -D feat/cmake_memap` Remove the old branch pointer.
 4. `git checkout -b feat/cmake_memap` Recreate the branch from the current commit.

screamerbg, 2018-03-23:
@vidavidorra Does this solve your issue?

vidavidorra, 2018-03-23:
@screamerbg So far yes, I'd like to see the `"-Wl,-Map=${CMAKE_PROJECT_NAME}.map"` in the profile, but that's not a huge deal for now because I use a custom profile at the moment. That seems to be a good solution to me, since it's real easy to create and use a custom profile.

I'm still waiting for ARMmbed/mbed-os#6254 to be merged, but only a little bit related to this issue. I'll close this than.

closed 2018-03-23 by vidavidorra