ARMmbed/mbed-cli #633: Should be able to specify profile with configuration file

lfdebrux, 2018-03-02:
I want to enable C++11 for a project.

Right now that means adding the `-std=c++11` flag to a toolchain profile, then adding the profile flag to `mbed compile` every time it is run.

Obviously this could be solved with an alias or a Makefile, but it would be cool if the profile could be saved in the `.mbed` file, like the target and toolchain, so it is remembered and optionally could be pushed to a repo.

Alternatively, would adding something to the mbed_app.json file be better? Maybe something like `target_overrides`, but called `profile_appends` or `toolchain_appends`.

I'm very happy to put together a pull request on this, I just wanted to know if this was a feature people were interested in/happy to accept.

theotherjimmy, 2018-03-02:
> Obviously this could be solved with an alias or a Makefile, but it would be cool if the profile could be saved in the .mbed file, like the target and toolchain, so it is remembered and optionally could be pushed to a repo.

That sounds great! @screamerbg This would help enable an easier path to reproduce-able builds for those who depend on a non-default profile. What do you think?

> Alternatively, would adding something to the mbed_app.json file be better? 

I'm also open to this. FYI, this would involve a PR to the mbed-os repo. 

--------

For reference, you may specify `--profile` multiple times. The JSON files will be merged for you.

lfdebrux, 2018-03-02:
> For reference, you may specify --profile multiple times. The JSON files will be merged for you.

Yeah, I noticed that! I created a repo https://github.com/RALSpaceASG/test-mbed-cxx11 where I was experimenting with how to use profiles.

> FYI, this would involve a PR to the mbed-os repo.

Ah good point.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*