ARMmbed/mbed-cli #635: Document lack of Cygwin support

donatieng, 2018-03-07:
Although we have a couple of issues open (#578, #299) referencing this, there is no mention in the docs that Cygwin is not supported:
https://os.mbed.com/docs/v5.7/tools/installation-and-setup.html

It would be great having either a mention in the docs, or (better!) the platform being detected at runtime with a warning such as "running mbed-cli within Cygwin, use at your own risks!".

cmonr, 2018-03-07:
At the very bottom of README.md, this line exists:

"Currently Mbed CLI is not compatible with Cygwin environment and cannot be executed inside it (https://github.com/ARMmbed/mbed-cli/issues/299)."

Are we saying that we don't expect users to read the README, or that we should add more references around the codebase explicitly stating the Cygwin isn't supported?

cmonr, 2018-03-07:
Another option would be to somehow trim or split up the current README. Imo, it does seem quite long.

@theotherjimmy @screamerbg

donatieng, 2018-03-07:
Yes, it's very far down!
And there is no mention of it in the handbook.

cmonr, 2018-03-08:
@sg- @AnotherButler Looping y'all in.

screamerbg, 2018-03-23:
@cmonr Splitting the README might lead to another problem - "It's in the documentation but I couldn't find it (Ctrl+F)". Perhaps it's a question of shortening some of the Mbed CLI readme content @AnotherButler 

Also agreed that this should be in the handbook @AnotherButler

AnotherButler, 2018-04-03:
This has been in the Handbook: https://os.mbed.com/docs/latest/tools/test-and-debug.html

screamerbg, 2018-04-28:
Closing this issue. Let's follow-up in #578 instead. In my environment Cygwin v2.10 (latest) for Windows 10 64bit, Mbed CLI works just fine.

closed 2018-04-28 by screamerbg