ARMmbed/mbed-cli #636: Makefiles produced by mbed export don't include full paths to toolchains

donatieng, 2018-03-07:
For instance, exporting with `mbed export -i MAKE_ARMC5 -m NUCLEO_F429ZI` will generate a Makefile calling `armcc` directly. If the compiler's location is not present in `PATH`, `make` will fail.

In that case the path configured in the `ARM_PATH` configuration variable should be used in the Makefile.

theotherjimmy, 2018-03-07:
@donatieng That's by design. What would we resolve those paths to in the online compiler's export?

theotherjimmy, 2018-03-07:
See [the handbook documentation](https://os.mbed.com/docs/v5.7/tools/exporting.html). The Make and Eclipse section calls out this requirement in a NOTE.

donatieng, 2018-03-07:
It's a feature not a bug then ;) thoughI find it counter-intuitive.

Most similar tools include an overridable variable at the start of the Makefile, e.g.:
`TOOLCHAIN_PATH?=/Users/dongar01/toolchains/gcc-arm-none-eabi-6-2017-q2-update/`.

This could be left empty for what's generated by the online compiler. But as a user, I don't care what the online compiler is using under the hood to export and would rather not have this limitation with the CLI.

theotherjimmy, 2018-03-07:
If you would like to see that change, could you open an issue on mbed-os?

screamerbg, 2018-03-23:
Bump @donatieng