ARMmbed/mbed-cli #638: Windows installer installs old version of Mbed CLI

dhwalters423, 2018-03-08:
The Windows installer linked here: https://os.mbed.com/docs/v5.7/tools/installing-with-the-windows-installer.html is installing version 1.2.2 rather than the newest version 1.4.

screamerbg, 2018-03-23:
@arekzaluski Is it possible to add an option/checkbox to the installer to `[ ] Install the latest Mbed CLI (requires Internet connection)` which effectively invokes `pip install -U mbed-cli` ?

arekzaluski, 2018-03-26:
@screamerbg  @dhwalters423 Yes, we already have a ticket to support this functionality. I don't know yet the estimated time of arrival for it.

screamerbg, 2018-04-28:
@arekzaluski Any update on this?

asifrizwanubx, 2018-07-11:
@screamerbg 

I prefer fast and easy installation (one click) and I used Windows installer but today I noticed it is old version of mbed-cli (mbed-cli 1.1.1) and pip install -U mbed-cli never work with it.

And due to which I faced issues in importing of mbed-os.

asifrizwanubx, 2018-07-11:
Error:

Collecting mbed-cli
  Using cached https://files.pythonhosted.org/packages/2e/b6/9c5112f682fbb0dc516e9e29cb16fcc7a9ce8482d0e7fe5c8d463948a313/mbed-cli-1.7.3.zip
Installing collected packages: mbed-cli
  Found existing installation: mbed-cli 1.1.1
    Uninstalling mbed-cli-1.1.1:
      Successfully uninstalled mbed-cli-1.1.1
  Running setup.py install for mbed-cli ... done
Exception:
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pip\basecommand.py", line 215, in main
    status = self.run(options, args)
  File "c:\python27\lib\site-packages\pip\commands\install.py", line 350, in run
    isolated=options.isolated_mode,
  File "c:\python27\lib\site-packages\pip\commands\install.py", line 436, in get_lib_location_guesses
    scheme = distutils_scheme('', *args, **kwargs)
  File "c:\python27\lib\site-packages\pip\locations.py", line 140, in distutils_scheme
    d = Distribution(dist_args)
  File "c:\python27\lib\site-packages\setuptools\dist.py", line 316, in __init__
    for ep in pkg_resources.iter_entry_points('distutils.setup_keywords'):
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 730, in iter_entry_points
    entries = dist.get_entry_map(group)
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 2639, in get_entry_map
    self._get_metadata('entry_points.txt'), self
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 2369, in parse_map
    for group, lines in data:
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 2967, in split_sections
    for line in yield_lines(s):
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 2235, in yield_lines
    for ss in strs:
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 2564, in _get_metadata
    if self.has_metadata(name):
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1462, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1822, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1702, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1642, in load
    mtime = os.stat(path).st_mtime
WindowsError: [Error 2] The system cannot find the file specified: 'c:\\python27\\lib\\site-packages\\mbed_cli-1.1.1-py2.7.egg'
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\pip.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\pip\__init__.py", line 233, in main
    return command.main(cmd_args)
  File "c:\python27\lib\site-packages\pip\basecommand.py", line 252, in main
    pip_version_check(session)
  File "c:\python27\lib\site-packages\pip\utils\outdated.py", line 102, in pip_version_check
    installed_version = get_installed_version("pip")
  File "c:\python27\lib\site-packages\pip\utils\__init__.py", line 838, in get_installed_version
    working_set = pkg_resources.WorkingSet()
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 644, in __init__
    self.add_entry(entry)
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 700, in add_entry
    for dist in find_distributions(entry, True):
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1949, in find_eggs_in_zip
    if metadata.has_metadata('PKG-INFO'):
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1463, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1823, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1703, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1643, in load
    mtime = os.stat(path).st_mtime
WindowsError: [Error 2] The system cannot find the file specified: 'c:\\python27\\lib\\site-packages\\mbed_cli-1.1.1-py2.7.egg'

arekzaluski, 2018-07-11:
@asifrizwanubx Thanks for reporting. We are looking into improving the installer. Allowing it to install the latest version of mbed-cli if internet connection is present and installing mbed-cli in a way that it can be easily updated.

Adding @theotherjimmy

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

dhwalters423, 2018-10-03:
@adbridge can you clarify what you need from me?

adbridge, 2018-10-03:
@dhwalters423 Apologies, the comment is from our new automated issue checking script that got rolled out yesterday. Unfortunately it looks like I missed some configuration in a number of repos. I will fix it :)

dhwalters423, 2018-10-17:
Hi @arekzaluski I just ran the installer on a fresh machine and it still installs an older version of mbed-cli. Most notably, the mbed cli version that is bundled does not include any of the new device management capabilties which is a really nice feature, and one that our new Pelion documentation references.

Can this be prioritized higher since we now have critical features listed in the documentation that are not available in this version?