ARMmbed/mbed-cli #645: mbed-cli gcc arm compile problem

JuricaResetar, 2018-03-16:
Hi,
I'm trying to use mbed-cli to work with mbed-os offline.

### Description
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.

**mbed-cli version:** 
1.4.0.

**mbed-os sha:**
0b37629 Merge pull request #114 from adbridge/master

**Actual behavior**
sudo mbed compile -t GCC_ARM -m NRF52_DK
Building project mbed-os-example-blinky (NRF52_DK, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Scan: FEATURE_BLE
Compile [  1.7%]: BusIn.cpp
[Fatal Error] platform.h@22,19: cstddef: No such file or directory
[ERROR] In file included from ./mbed-os/drivers/BusIn.h:19:0,
                 from ./mbed-os/drivers/BusIn.cpp:16:
./mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory
 #include <cstddef>
                   ^
compilation terminated.
[ERROR]: /usr/bin/python" returned error code 1.
[ERROR]: Command "/usr/bin/python -u /home/simvelop/mbedOS/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NRF52_DK --source . --build ./BUILD/NRF52_DK/GCC_ARM" in "/home/simvelop/mbedOS/mbed-os-example-blinky"

-----------------------------------------------------------------

## Question
How to setup my toolchain to solve the problem? I've seen the same problems and solution with updating gcc arm compiler but I would realy like to avoid that beacuse of some other requirement I need to meet. 
Here's a content of my mbed_settings.py file:
```
from os.path import join, abspath, dirname

#ROOT = abspath(join(dirname(__file__), "."))

##############################################################################
# Build System Settings
##############################################################################
#BUILD_DIR = abspath(join(ROOT, "build"))

# ARM
#ARM_PATH = "C:/Program Files/ARM"

# GCC ARM
GCC_ARM_PATH = "/usr/bin/"

# GCC CodeRed
#GCC_CR_PATH = "C:/code_red/RedSuite_4.2.0_349/redsuite/Tools/bin"

# IAR
#IAR_PATH = "C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.0/arm"

# Goanna static analyser. Please overload it in private_settings.py
#GOANNA_PATH = "c:/Program Files (x86)/RedLizards/Goanna Central 3.2.3/bin"

#BUILD_OPTIONS = []

# mbed.org username
#MBED_ORG_USER = ""
```

screamerbg, 2018-03-16:
cc @theotherjimmy @BlackstoneEngineering

JuricaResetar, 2018-03-17:
The solution for my problem was to install arm-none-eabi-newlib package.
Please see more info here #6381.

closed 2018-03-17 by JuricaResetar