ARMmbed/mbed-cli #646: new --create-only does not download mbed tools

michaelVisimid, 2018-03-20:
```
C:\STWorkspace>py -2.7 -m mbed new v_05_test_3 --create-only
[mbed] Creating new program "v_05_test_3" (git)
[mbed] WARNING: Cannot find the mbed tools directory in "C:\STWorkspace\v_05_test_3"
---

C:\STWorkspace\v_05_test_3>py -2.7 -m mbed add http://os.mbed.com/users/mbed_official/code/mbed-dev/
[mbed] Adding library "mbed-dev" from "https://os.mbed.com/users/mbed_official/code/mbed-dev" at latest revision in the current branch
abort: data/targets/TARGET_ARM_SSG/TARGET_BEETLE/PeripheralNames.h.i@3d7f101b08f6: no match found!
```
Tools copied into the directory manually at this point.

```
C:\STWorkspace\v_05_test_3>py -2.7 -m mbed add http://os.mbed.com/users/mbed_official/code/mbed-dev/
[mbed] Adding library "mbed-dev" from "https://os.mbed.com/users/mbed_official/code/mbed-dev" at latest revision in the current branch
warning: LF will be replaced by CRLF in mbed-dev.lib.
The file will have its original line endings in your working directory.

C:\STWorkspace\v_05_test_3>py -2.7 -m mbed export -i SW4STM32 -m NUCLEO_L073RZ
Using targets from C:\STWorkspace\v_05_test_3\mbed-dev\targets\targets.json
Scan: .

Create a System Workbench for STM32 managed project
Project name: v_05_test_3
Target:       NUCLEO_L073RZ
Toolchain:    GCC_ARM

Symbols: 47
Include folders: 16
Source folders: 9, with 38 exclusions
Exclude folders: 38
Linker script:   mbed-dev/targets/TARGET_STM/TARGET_STM32L0/TARGET_NUCLEO_L073RZ/device/TOOLCHAIN_GCC_ARM/STM32L073XZ.ld

C:\STWorkspace\v_05_test_3>
```

theotherjimmy, 2018-03-20:
@michaelVisimid I think `--create-only` is supposed to not download the tools. @screamerbg Could you confirm that this is expected behavior?

michaelVisimid, 2018-03-20:
This is a new change between v4 and 5. In the previous version --create-only downloaded the tools. If it doesn't download the tools libraries cannot be added, program exported, etc. I'm guessing this was not the intended behavior, and if it is my question would be why?

theotherjimmy, 2018-03-21:
@michaelVisimid I think you're right that it's a new behavior. When you say "v4 and v5" what are you referring to?

michaelVisimid, 2018-03-22:
Sorry I was a little distracted when I wrote that. I meant 1.5.0 and 1.4.0. The latest version that pip installs and the previous version that pip installed.

screamerbg, 2018-03-23:
@michaelVisimid Yes, that's correct. Starting with Mbed CLI 1.5.0, the Mbed 2 tools are downloaded only if Mbed 2/Classic library is in the source tree. The purpose of these tools is to support Mbed 2/Classic releases. Previously (< v.1.5.0) Mbed CLI would download the Mbed 2 tools tools if even if you import a repository that doesn't contain neither Mbed OS 5 nor Mbed 2/Classic library. This could be very confusing for when you simply want repositories downloaded, no tools, no "magic". This is the usecase/workflow/issue we wanted to address.

michaelVisimid, 2018-03-23:
That makes sense. Thank you for the clarification.

screamerbg, 2018-03-26:
@michaelVisimid are you happy to close the ticket if resolved?

michaelVisimid, 2018-03-27:
I think closing the ticket is the correct action.

closed 2018-05-26 by screamerbg