ARMmbed/mbed-cli #647: new --mbedlib downloads mbed libraries in unusable format for export to SW4STM32

michaelVisimid, 2018-03-20:
The fix to the mbedlib not being downloaded on creation is using a directory structure which does not export to SW4STM32 correctly. I have not tested other IDEs.

Project creation and compilation:

```
C:\STWorkspace>py -2.7 -m mbed new v_05_test --mbedlib
[mbed] Creating new program "v_05_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Unpacking library build "aa5281ff4a02" in "C:\STWorkspace\v_05_test\mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/aa5281ff4a02"
warning: LF will be replaced by CRLF in mbed.bld.
The file will have its original line endings in your working directory.
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

C:\STWorkspace\v_05_test>py -2.7 -m mbed compile -t GCC_ARM -m NUCLEO_L073RZ
Building project v_05_test (NUCLEO_L073RZ, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [100.0%]: main.cpp
Link: v_05_test
Elf2Bin: v_05_test
+---------------------------------------------------------------------------------+-------+-------+------+
| Module                                                                          | .text | .data | .bss |
+---------------------------------------------------------------------------------+-------+-------+------+
| BUILD\NUCLEO_L073RZ\GCC_ARM\main.o                                              |    64 |     4 |   28 |
| [fill]                                                                          |    28 |     0 |    5 |
| [lib]/c.a                                                                       | 26539 |  2472 |   89 |
| [lib]/gcc.a                                                                     |  9456 |     0 |    0 |
| [lib]/mbed.a                                                                    |  1780 |     4 |   70 |
| [lib]/misc                                                                      |   220 |     8 |   28 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\PeripheralPins.o       |   276 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\gpio_api.o             |   498 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\hal_tick_16b.o         |   328 |     0 |    4 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\mbed_board.o           |   202 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\mbed_overrides.o       |    20 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\mbed_retarget.o        |   662 |     4 |  264 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\mbed_sdk_boot.o        |    68 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\pinmap.o               |   562 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\serial_api.o           |  1005 |     0 |   80 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\serial_device.o        |    72 |     0 |  560 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\startup_stm32l073xx.o  |   246 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\stm32l0xx_hal.o        |    62 |     0 |    4 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\stm32l0xx_hal_rcc.o    |  1824 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\stm32l0xx_hal_rcc_ex.o |   416 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\stm32l0xx_hal_tim.o    |   276 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\stm32l0xx_hal_uart.o   |  1095 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\system_clock.o         |   432 |     0 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\system_stm32l0xx.o     |   173 |     4 |    0 |
| mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\us_ticker_16b.o        |   202 |     0 |   64 |
| Subtotals                                                                       | 46506 |  2496 | 1196 |
+---------------------------------------------------------------------------------+-------+-------+------+
Total Static RAM memory (data + bss): 3692 bytes
Total Flash memory (text + data): 49002 bytes

Image: .\BUILD\NUCLEO_L073RZ\GCC_ARM\v_05_test.bin

C:\STWorkspace\v_05_test>py -2.7 -m mbed export -i SW4STM32 -m NUCLEO_L073RZ
Using targets from C:\STWorkspace\v_05_test\mbed\aa5281ff4a02\targets.json
Scan: .

Create a System Workbench for STM32 managed project
Project name: v_05_test
Target:       NUCLEO_L073RZ
Toolchain:    GCC_ARM

Symbols: 47
Include folders: 14
Source folders: 1, with 164 exclusions
Exclude folders: 164
[... 233 lines cut ...]
Project name: v_05_test
Target:       NUCLEO_L073RZ
Toolchain:    GCC_ARM

Symbols: 47
Include folders: 16
Source folders: 10, with 39 exclusions
Exclude folders: 39
Linker script:   mbed-dev/targets/TARGET_STM/TARGET_STM32L0/TARGET_NUCLEO_L073RZ/device/TOOLCHAIN_GCC_ARM/STM32L073XZ.ld
```
Open project and it successfully builds:
```
make --no-print-directory post-build
Generating binary and Printing size information:
arm-none-eabi-objcopy -O binary "v_05_test-debug.elf" "v_05_test.bin"
arm-none-eabi-size -B "v_05_test-debug.elf"
   text	   data	    bss	    dec	    hex	filename
  22448	   2148	   1296	  25892	   6524	v_05_test-debug.elf
make ldclean
rm -rf linker-script-debug.ld
rm -rf linker-script-release.ld
 

13:37:23 Build Finished (took 2m:52s.578ms)
```

theotherjimmy, 2018-03-20:
@michaelVisimid Could you provide more detail about the error you encountered. I would really like to see the text of the error itself.

michaelVisimid, 2018-03-20:
No problem. I've tested the export with Keil and it seems to work no problem. There seems to be something wrong with the exporter and the precompilied mbed libraries.

This path doesn't seem to exist:

Path in question:
```
c:/ac6/systemworkbench/plugins/fr.ac6.mcu.externaltools.arm-none.win32_1.15.0.201708311556/tools/compiler/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/thumb/v6-m\libc.a

```

Path where libc.a resides:
```
C:\Ac6\SystemWorkbench\plugins\fr.ac6.mcu.externaltools.arm-none.win32_1.15.0.201708311556\tools\compiler\arm-none-eabi\lib\thumb\v6-m
```
If we goto 
```
C:\Ac6\SystemWorkbench\plugins\fr.ac6.mcu.externaltools.arm-none.win32_1.15.0.201708311556\tools\compiler\lib\gcc\arm-none-eabi\6.3.1\thumb\v6-m
```
We find libgcc.a

I am a bit busy so I don't have too much more time to look into this. Hopefully this will be enough info to point things in the right direction. I will check in tomorrow and provide more info if needed.

At this point my primary confusion is why mbed-OS programs created with the CLI or imported then exported with the CLI include the actual code whereas --mbedlib includes the precompilied libraries. Wouldn't it make sense to have the mbedlib program creation use the source code instead of the precompilied libraries as is the case with mbed-OS programs?

Summary of noticed issues:

* Precompilied mbed libraries will not export correctly to SW4STM32 - they seem to with Keil

* CLI import using mbed add http://os.mbed.com/users/mbed_official/code/mbed/ is not working

* Exported mbed-OS programs never use precompilied libraries whereas mbedlib programs do. Would it make more sense for mbedlib programs to use mbed-dev instead of the precompilied mbed libraries? The mbed libraries export with a terrible directory structure (revision ID is the main folder) in addition to having issues with GCC used by SW4STM32.

Here is the output from compilation of a program created using --mbedlib flag:
```
14:56:32 **** Build of configuration Debug for project v_05_test ****
make pre-build main-build 
Creating makefile.defs:
echo "export PREPROC_CMD = C:\Ac6\SystemWorkbench\plugins\fr.ac6.mcu.externaltools.arm-none.win32_1.15.0.201708311556\tools\compiler\bin/arm-none-eabi-cpp -E -P" > C:/STWorkspace/v_05_test/makefile.defs
 
Building file: ../main.cpp
Invoking: MCU G++ Compiler
C:\STWorkspace\v_05_test\Debug
arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -mfloat-abi=soft '-D__MBED__=1' '-DDEVICE_I2CSLAVE=1' -DTARGET_LIKE_MBED '-DDEVICE_PORTOUT=1' '-DDEVICE_PORTINOUT=1' -D__MBED_CMSIS_RTOS_CM -DTARGET_FF_MORPHO '-DMBED_BUILD_TIMESTAMP=1521575741.02' '-DDEVICE_RTC=1' -DTOOLCHAIN_object '-DDEVICE_SERIAL_ASYNCH=1' -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DTARGET_STM32L0 -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_M0P '-DDEVICE_ANALOGOUT=1' -DTARGET_UVISOR_UNSUPPORTED '-DDEVICE_SPI_ASYNCH=1' -DTARGET_NUCLEO_L073RZ '-DDEVICE_INTERRUPTIN=1' -DTARGET_CORTEX '-DDEVICE_I2C=1' '-DTRANSACTION_QUEUE_SIZE_SPI=2' -D__CORTEX_M0PLUS '-DDEVICE_STDIO_MESSAGES=1' '-DDEVICE_PWMOUT=1' -DTARGET_FAMILY_STM32 '-DDEVICE_LOWPOWERTIMER=1' -DTARGET_FF_ARDUINO -DTARGET_STM '-DDEVICE_SERIAL_FC=1' -DTARGET_STM32L073xx '-DDEVICE_TRNG=1' -DARM_MATH_CM0PLUS '-DDEVICE_PORTIN=1' '-DDEVICE_SLEEP=1' -DTOOLCHAIN_GCC_ARM '-DDEVICE_SPI=1' '-DDEVICE_SPISLAVE=1' '-DDEVICE_ANALOGIN=1' '-DDEVICE_SERIAL=1' -DTARGET_STM32L073RZ '-DDEVICE_FLASH=1' '-DDEVICE_I2C_ASYNCH=1' -DTARGET_LIKE_CORTEX_M0 -DMBED_DEBUG '-DMBED_TRAP_ERRORS_ENABLED=1' -DMBED_DEBUG '-DMBED_TRAP_ERRORS_ENABLED=1' -DNDEBUG -DNDEBUG -I"C:/STWorkspace/v_05_test" -I"C:/STWorkspace/v_05_test/mbed" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/drivers" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/hal" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/platform" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TARGET_STM" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TARGET_STM/TARGET_STM32L0" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TARGET_STM/TARGET_STM32L0/device" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TARGET_STM/TARGET_STM32L0/TARGET_NUCLEO_L073RZ" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TARGET_STM/TARGET_STM32L0/TARGET_NUCLEO_L073RZ/device" -I"C:/STWorkspace/v_05_test/mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TOOLCHAIN_GCC_ARM" -I"C:/STWorkspace/v_05_test"  -includeC:/STWorkspace/v_05_test/./mbed_config.h -O0 -funsigned-char -fno-delete-null-pointer-checks -fomit-frame-pointer -fmessage-length=0 -fno-builtin -g3 -Wall -Wextra -Wvla -Wno-unused-parameter -Wno-missing-field-initializers -ffunction-sections -fdata-sections -c -fno-exceptions -fno-rtti -ffunction-sections -MMD -MP -MF"main.d" -MT"main.o" -o "main.o" "../main.cpp"
Finished building: ../main.cpp
 
arm-none-eabi-cpp -E -P -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -Wl,-n -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group ../mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TOOLCHAIN_GCC_ARM/STM32L073XZ.ld -o linker-script-debug.ld
Building target: v_05_test-debug.elf
Invoking: MCU G++ Linker
arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -mfloat-abi=soft -L"C:\STWorkspace\v_05_test\mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM" -Wl,--wrap=main -Wl,--wrap=_malloc_r -Wl,--wrap=_free_r -Wl,--wrap=_realloc_r -Wl,--wrap=_memalign_r -Wl,--wrap=_calloc_r - [... cut ...]
Using targets from C:\STWorkspace\mbed_blinky\mbed\aa5281ff4a02\targets.json
Scan: .
Warning: Multiple linker scripts detected: .\Debug\linker-script-debug.ld -> .\mbed\aa5281ff4a02\TARGET_NUCLEO_L073RZ\TOOLCHAIN_GCC_ARM\STM32L073XZ.ld

Create a System Workbench for STM32 managed project
Project name: mbed_blinky
Target:       NUCLEO_L073RZ
Toolchain:    GCC_ARM

Symbols: 47
Include folders: 19
Source folders: 1, with 165 exclusions
Exclude folders: 165
Linker script:   mbed/aa5281ff4a02/TARGET_NUCLEO_L073RZ/TOOLCHAIN_GCC_ARM/STM32L073XZ.ld
```

theotherjimmy, 2018-03-21:
@michaelVisimid Thanks for the logs.

Some comments for later: 
 * It looks like the exporter is adding the `-Wl,--wrap__malloc` when it should not be

-------

> Wouldn't it make sense to have the mbedlib program creation use the source code instead of the precompilied libraries as is the case with mbed-OS programs?

This is a legacy thing: mbedlib program creation corresponds to Mbed 2, which has always been released as a pre-compiled library. Mbed OS generally refers to Mbed OS 5, which has always been released as source code so that you may configure it using the config system. I hope that answers your question

screamerbg, 2018-05-26:
Closing due to inactivity. Please reopen if this is still an issue.

closed 2018-05-26 by screamerbg