ARMmbed/mbed-cli #648: Generate a default mbed_app.json with "mbed new" to increase config system awareness

loverdeg-ep, 2018-03-21:
See [handbook-#462](https://github.com/ARMmbed/Handbook/issues/462#issuecomment-374974978)

It's also one less step the user needs to complete themselves. Step being, lookup name of file to create, mbed_app.json, and the basic syntax to follow.

loverdeg-ep, 2018-04-10:
@theotherjimmy does this belong in mbedOS?

Also, I'm finally starting to lock into my brain the names of mbed_app.json, mbed_lib.json, and the json object sections that go within them. All the previous syntaxes for yotta and whatever else was used since then weren't helping.

However I still consistently need to look up the names of sections.
This enhancement would save me a lot of time, confusion and allow me to onboard new developers more quickly.

theotherjimmy, 2018-04-10:
@loverdeg-ep This belongs in Mbed CLI.

>This enhancement would save me a lot of time, confusion and allow me to onboard new developers more quickly.

Agreed, this is probably a good idea.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

PR #818 (2019-01-30): Modify "mbed new" to create mbed_app.json with default content.
Modify "mbed new" to create mbed_app.json with default content.

Fixes #648