ARMmbed/mbed-cli #653: Progress bar only goes up to 90%

LiyouZhou, 2018-04-04:
v 1.5.0
```
mbed add https://github.com/ARMmbed/mbed-bootloader-internal
[mbed] Adding library "mbed-bootloader-internal" from "ssh://git@github.com/ARMmbed/mbed-bootloader-internal.git" at latest revision in the current branch
Downloading |########--------------------------------------------------| 14.58
```

The progress bar above only ever goes up to 90.0% and it disappears at 90.0%.

theotherjimmy, 2018-04-04:
@LiyouZhou I think that should be fine, so long as the operation is complete after the progress bar goes away.

theotherjimmy, 2018-04-04:
It looks like the last bit is "checking out files" https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L961-L962

@screamerbg This may be a bug with the parsing of that progress.

screamerbg, 2018-04-25:
@LiyouZhou What version of Git do you use?

screamerbg, 2018-04-28:
Just tried it on Git 1.9.5 (Windows), Git 2.2.0 (Windows) and Git 2.10.1 (standard Git on Mac). On all of these I get progress bar between 90% and 100%. @LiyouZhou Please let me know of you Git version so I can reproduce.

screamerbg, 2018-04-28:
@LiyouZhou Also could you check the behavior when you import mbed-os? E.g.
```
$ mbed cache purge
$ mbed new test_program
```

screamerbg, 2018-05-17:
Bump @LiyouZhou

closed 2018-06-06 by LiyouZhou