ARMmbed/mbed-cli #654: Installation may fail even with sudo due to lack of admin privileges

linlingao, 2018-04-04:
While following the example to create a program documented here: 
https://os-doc-builder.test.mbed.com/docs/development/tools/working-with-mbed-cli.html
I ran into some warnings:
bash-3.2$ python mbed.py new mbed-os-program
[mbed] Creating new program "mbed-os-program" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#16bac101a6b7b4724023dcf86ece1548e3a23cbf"
[mbed] Auto-installing missing Python modules...
[mbed] WARNING: Unable to auto-install required Python modules.
—
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed OS tools in this program require the following Python modules: mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy, pyelftools
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "/Users/lingao01/mbed-cli/mbed/mbed-os-program/mbed-os"
[mbed] WARNING: On Posix systems (Linux, Mac, etc) you might have to switch to superuser account or use "sudo"

I then used sudo to execute "pip install -r requirements.txt", the following error occurred:
Collecting setuptools>=19.0 (from hidapi->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
  Using cached setuptools-39.0.1-py2.py3-none-any.whl
Installing collected packages: setuptools, hidapi, pyOCD, mbed-host-tests, lockfile, funcsigs, pbr, mock, mbed-greentea, beautifulsoup4, fuzzywuzzy, pyelftools
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 640, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/utils/_init_.py", line 315, in renames
    shutil.move(old, new)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors

It doesn't seem "sudo" works. I finally ran "Make Me Local Admin" app and succeeded. This should be documented.

theotherjimmy, 2018-04-04:
@screamerbg I think the recommended use of pip on mac may be to add `--user`. We can document this @linlingao.

screamerbg, 2018-04-28:
@linlingao Thanks for the detailed investigation. Do you think you could send a PR to the documentation so this is mentioned?

PR #763 (2018-09-27): Correct pip install help message on Mac
Fixes #654
commit e89c68a8 (2018-09-27): Correct pip install help message on Mac

Fixes #654
closed 2018-09-27 by screamerbg