ARMmbed/mbed-cli #657: pip 10 support - no attribute get_installed_distributions

alzix, 2018-04-15:
it seems that `pip.get_installed_distributions()` is no longer available in pip 10.

`get_installed_distributions` is reference from https://github.com/ARMmbed/mbed-cli/blob/53f8b113c9b00ecffbf6031cc4b34333b8f52cf7/mbed/mbed.py#L1452
`mbed compile` fails with the following error:
```
$ mbed compile
[mbed] ERROR: Unknown Error: 'module' object has no attribute 'get_installed_distributions'
```

hug-dev, 2018-04-16:
(`python -m pip install pip==9.0.3` to go back to the old version of `pip` that works with `mbed-cli`, safer in a virtual environment, until this gets fixed)

fkjagodzinski, 2018-04-16:
I've also run into this bug today and found one thing worth sharing before finding this GitHub issue: https://pip.pypa.io/en/stable/user_guide/#using-pip-from-your-program -- **pip's internal API was never considered stable and should not be used here**. The fix should use other Python module or call pip via the terminal.

theotherjimmy, 2018-04-16:
I was looking for that text the other day and could not find it. Thanks @fkjagodzinski

theotherjimmy, 2018-04-16:
Based on the link, importing pip is a bug.

JanneKiiskila, 2018-04-19:
This needs to be fixed ASAP, because `pip` will nag about this! I did it now, but now stuff is broken and I can't compile anymore. However, downgrading `pip` did not work, either. 

```
jankii01@ubuntu:~/mbed/mbed-os-example-blinky$ python -m pip install pip==9.0.3
Collecting pip==9.0.3
From cffi callback <function _verify_callback at 0x7f949f914668>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f949a705488>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f949a705d70>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f949a705ed8>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f949a6d30c8>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f949a6d3230>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Could not fetch URL https://pypi.python.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)) - skipping
  Could not find a version that satisfies the requirement pip==9.0.3 (from versions: )
No matching distribution found for pip==9.0.3
jankii01@ubuntu:~/mbed/mbed-os-example-blinky$ python -m pip install pip==9.1.0
Collecting pip==9.1.0
From cffi callback <function _verify_callback at 0x7f235d140668>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f2357f31488>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f2357f31d70>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)': /simple/pip/
From cffi callback <function _verify_callback at 0x7f2357f31ed8>:
Traceback (most recent call last):
[... cut ...]
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Could not fetch URL https://pypi.python.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)) - skipping
  Could not find a version that satisfies the requirement pip==9.1.0 (from versions: )
No matching distribution found for pip==9.1.0
```

JanneKiiskila, 2018-04-19:
Seems I have another package mismatch then as well...

jeromecoutant, 2018-04-25:
Hi
Any news about this pip issue ?  :-(
Thx

screamerbg, 2018-04-25:
Just put up a PR for this, which will fix future compatibility as well. A new version of Mbed CLI will be released to addresses this by the end of today (25th of April)

JanneKiiskila, 2018-04-25:
And you mean this PR; https://github.com/ARMmbed/mbed-cli/pull/661

screamerbg, 2018-04-27:
The Mbed CLI 1.5.1 is now live and addresses this issue. @alzix @JanneKiiskila @hug-dev thanks for the bug report

JanneKiiskila, 2018-04-27:
Thanks for quick fix!

closed 2018-04-28 by screamerbg

WonderMr, 2018-05-25:
For pip 10 use `from pip._internal.utils.misc import get_installed_distributions`

theotherjimmy, 2018-05-25:
@WonderMr That's pretty internal sounding. We don't want to rely on internals.

xvart, 2018-10-01:
So had this awesome error today on ubuntu 18, and one of the many reasons I will never use python in production the thing is a never ending moving target of crazy

richdevboston, 2019-08-13:
yes, this is pretty frustrating.  Never ending moving target seems right with many of the system-level utilities like pip and conda.  Pip especially.