ARMmbed/mbed-cli #663: Netbeans export incomplete causing compilation failure

dellgreen, 2018-04-27:
When exporting mbed-os-example-blinky example to netbeans, the netbeans configurations.xml does not contain the full path to the C preprocessor ('arm-none-eabi-cpp' in my case). All other tools in this xml file are absolute and contain the full path.

Correcting manually the configurations.xml file before opening in netbeans fixes the problem

https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/

dellgreen, 2018-04-27:
no, using command line tools

theotherjimmy, 2018-04-27:
@dellgreen The code that implements the exporters is [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/nb/__init__.py) I'll get time to look at this next week.

dellgreen, 2018-04-27:
ok great, many thanks :)

theotherjimmy, 2018-04-30:
This looks intentional: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/nb/__init__.py#L205-L208

@cmens23 Could you comment on this?

dellgreen, 2018-05-25:
anyone else know why this may be?

theotherjimmy, 2018-09-24:
No comments from the original author. My vote is to make this consistent across tool invocations.