ARMmbed/mbed-cli #671: mbed import issue

oneway3124, 2018-05-16:
C:\Users\dav>mbed --version
1.5.1

C:\Users\dav\Desktop>mbed import https://os.mbed.com/teams/mbed/code/mbed_blinky
/
[mbed] Importing program "mbed_blinky" from "https://os.mbed.com/teams/mbed/code
/mbed_blinky" at latest revision in the current branch

[mbed] Adding library "mbed" from "https://os.mbed.com/users/mbed_official/code/
mbed/builds" at rev #aa5281ff4a02
[mbed] Downloading library build "aa5281ff4a02" (might take a minute)

But it cannot be executed successfully.
I want to know how to fix it. Thanks a lot in advance if you can give me some help.

oneway3124, 2018-05-16:
I think that this issue is an network issue, because this package of 300MB can be download by me quickly.
So the mbed.py create some error.

cmonr, 2018-05-21:
@oneway3124 Could you post the output of the command when you run it with `-vv` ?

oneway3124, 2018-05-22:
@cmonr I has tried to use your github code, as the following
https://github.com/cmonr/mbed-os-example-wifi
[image]
Is it ok?

cmonr, 2018-05-22:
@oneway3124 `-vv` is meant to be appended to the command as an additional parameter.

So instead of running `mbed import mbed-os-example-wifi`, the command would be `mbed -vv import mbed-os-example-wifi`

oneway3124, 2018-05-24:
I has done successfully now.

closed 2018-05-24 by oneway3124