ARMmbed/mbed-cli #677: v1.7.0 is broken on Python 2

amq, 2018-05-25:
When `v1.7.0` got released today, my CI builds started to fail with:

```
Traceback (most recent call last):
  File "/usr/local/bin/mbed", line 7, in <module>
    from mbed.mbed import main
  File "/usr/local/lib/python2.7/dist-packages/mbed/mbed.py", line 22, in <module>
    from future.builtins.iterators import zip
ImportError: No module named future.builtins.iterators
```

`Dockerfile` to replicate the issue:

```
FROM debian:stretch-slim
RUN apt update && \
    apt install -y lbzip2 ca-certificates wget python-pip git mercurial --no-install-suggests && \
    rm -rf /var/lib/apt/lists/*
RUN pip install mbed-cli
RUN mbed config -G GCC_ARM_PATH "/usr/bin"
```

amq, 2018-05-25:
Strangely, Python 3 also doesn't seem to work:

```
Traceback (most recent call last):
  File "/usr/local/bin/mbed", line 7, in <module>
    from mbed.mbed import main
  File "/usr/local/lib/python3.5/dist-packages/mbed/mbed.py", line 22, in <module>
    from future.builtins.iterators import zip
ImportError: No module named 'future'
```

```
FROM debian:stretch-slim
RUN apt update && \
    apt install -y lbzip2 ca-certificates wget python3-pip git mercurial --no-install-suggests && \
    rm -rf /var/lib/apt/lists/*
RUN pip3 install mbed-cli
RUN mbed config -G GCC_ARM_PATH "/usr/bin"
```

theotherjimmy, 2018-05-25:
@amq It looks like we're depending on the `future` package in pip, but it was not added to the dependencies in setup.py.

theotherjimmy, 2018-05-25:
As a workaround you should be able to pip install future.

amq, 2018-05-25:
Installing future fixes the problem.

amq, 2018-05-25:
There is a different problem when building:

```
Traceback (most recent call last):
  File "/home/amq/project/mbed-os/tools/make.py", line 41, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/home/amq/project/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/home/amq/project/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/home/amq/project/mbed-os/tools/targets/__init__.py", line 555, in <module>
    update_target_data()
  File "/home/amq/project/mbed-os/tools/targets/__init__.py", line 544, in update_target_data
    in Target.get_json_target_data().items()
  File "/home/amq/project/mbed-os/tools/targets/__init__.py", line 70, in wrapper
    CACHES[(func.__name__, args)] = func(*args, **kwargs)
  File "/home/amq/project/mbed-os/tools/targets/__init__.py", line 142, in get_json_target_data
    Target.__targets_json_location_default)
  File "/home/amq/project/mbed-os/tools/utils.py", line 351, in json_file_to_dict
    object_pairs_hook=OrderedDict)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
[mbed] ERROR: "/usr/bin/python3" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python3 -u /home/amq/project/mbed-os/tools/make.py -t GCC_ARM -m REDACTED --source ./bootloader --source ./mbed-os --source ./lib/flashiap-driver --build ./BUILD/REDACTED/GCC_ARM -c" in "/home/amq/project"
```

screamerbg, 2018-05-26:
cc @cmonr

screamerbg, 2018-05-26:
@amq Thanks for the bug report. I've immediately reproduced the issue by removing `future`. I've also reissued a hotfix - #678 and published the Mbed CLI 1.7.1 release. Could you please update your Mbed CLI to 1.7.1 and let me know whether Python 2 is still broken for you?

amq, 2018-05-26:
Python 2 is fine now.

Python 3 problem remains.

cmonr, 2018-05-26:
@amq Interesting. Looks like there are still some tools in mbed-os that I missed.

I'll fix the issue and make a commit in the mbed-os repo, and ask if you can verify the fix.

cmonr, 2018-05-26:
Closing this here, and moving this to https://github.com/ARMmbed/mbed-os/issues/7026 since the issue is now focused around an mbed-os tool.

closed 2018-05-26 by cmonr