ARMmbed/mbed-cli #683: mbed-cli complains about python package requirements when they are already satisfied...

ansondtx20, 2018-05-30:
On my Mac, running 10.13.4 with brew installed, I am unable to convince mbed-cli v1.7.1 that the prerequisite packages needed (and called out in mbed-os/requirements.txt) are all installed and available.. each invocation of mbed-cli results in mbed-cli trying to (and failing for some unknown reason) to rectify those requirements...

I have installed python 2.7.15 via brew and access it in /usr/local/bin.  pip is at 9.0.3 and is also installed in /usr/local/bin. mbed-cli is also installed in /usr/local/bin and was installed via pip 9.0.3

Here is a sample dump of what I am seeing. Thanks!:

mac% mbed --version
1.7.1
mac% mbed compile -j0 -m K64F
[mbed] Auto-installing missing Python modules...
[mbed] WARNING: Unable to auto-install required Python modules.
---
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed OS tools in this program require the following Python modules: mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy, pyelftools, jsonschema
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "/Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample/mbed-os"
[mbed] WARNING: On Posix systems (Linux, Mac, etc) you might have to switch to superuser account or use "sudo"
---
Traceback (most recent call last):
  File "/Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample/mbed-os/tools/make.py", line 48, in <module>
    from tools.build_api import build_project
  File "/Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample/mbed-os/tools/build_api.py", line 36, in <module>
    from .arm_pack_manager import Cache
  File "/Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample/mbed-os/tools/arm_pack_manager/__init__.py", line 5, in <module>
    from bs4 import BeautifulSoup
ImportError: No module named bs4
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -j0" in "/Users/douans01/Documents/workspace/GitHub/morpheus_projects/build/mbed-cloud-sample"
---
mac% cd mbed-os
mac% sudo pip install -Ur requirements.txt
Password:
The directory '/Users/douans01/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/douans01/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: colorama>=0.3.3 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 1))
Requirement already up-to-date: PySerial>=2.7 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 2))
Requirement already up-to-date: PrettyTable>=0.7.2 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 3))
Requirement already up-to-date: Jinja2>=2.7.3 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 4))
Requirement already up-to-date: IntelHex>=1.3 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 5))
Requirement already up-to-date: junit-xml in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 6))
Requirement already up-to-date: pyYAML in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 7))
Requirement already up-to-date: requests in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 8))
Requirement already up-to-date: mbed-ls>=0.2.13 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 9))
Requirement already up-to-date: mbed-host-tests>=1.1.2 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 10))
Requirement already up-to-date: mbed-greentea>=0.2.24 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 11))
Requirement already up-to-date: beautifulsoup4>=4 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 12))
Requirement already up-to-date: fuzzywuzzy>=0.11 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 13))
Requirement already up-to-date: pyelftools>=0.24 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 14))
Requirement already up-to-date: jsonschema>=2.6 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 15))
Requirement already up-to-date: future>=0.16.0 in /usr/local/lib/python2.7/site-packages (from -r requirements.txt (line 16))
Requirement already up-to-date: MarkupSafe>=0.23 in /usr/local/lib/python2.7/site-packages (from Jinja2>=2.7.3->-r requirements.txt (line 4))
Requirement already up-to-date: six in /usr/local/lib/python2.7/site-packages (from junit-xml->-r requirements.txt (line 6))
Requirement already up-to-date: certifi>=2017.4.17 in /usr/local/lib/python2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already up-to-date: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already up-to-date: idna<2.7,>=2.5 in /usr/local/lib/python2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already up-to-date: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already up-to-date: fasteners in /usr/local/lib/python2.7/site-packages (from mbed-ls>=0.2.13->-r requirements.txt (line 9))
Requirement already up-to-date: appdirs>=1.4 in /usr/local/lib/python2.7/site-packages (from mbed-ls>=0.2.13->-r requirements.txt (line 9))
Requirement already up-to-date: pyOCD>=0.8.1a1 in /usr/local/lib/python2.7/site-packages (from mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: lockfile in /usr/local/lib/python2.7/site-packages (from mbed-greentea>=0.2.24->-r requirements.txt (line 11))
Requirement already up-to-date: mock in /usr/local/lib/python2.7/site-packages (from mbed-greentea>=0.2.24->-r requirements.txt (line 11))
Requirement already up-to-date: functools32; python_version == "2.7" in /usr/local/lib/python2.7/site-packages (from jsonschema>=2.6->-r requirements.txt (line 15))
Requirement already up-to-date: monotonic>=0.1 in /usr/local/lib/python2.7/site-packages (from fasteners->mbed-ls>=0.2.13->-r requirements.txt (line 9))
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: websocket-client in /usr/local/lib/python2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: intervaltree in /usr/local/lib/python2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: hidapi in /usr/local/lib/python2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: pbr>=0.11 in /usr/local/lib/python2.7/site-packages (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 11))
Requirement already up-to-date: funcsigs>=1; python_version < "3.3" in /usr/local/lib/python2.7/site-packages (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 11))
Requirement already up-to-date: sortedcontainers in /usr/local/lib/python2.7/site-packages (from intervaltree->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already up-to-date: setuptools>=19.0 in /usr/local/lib/python2.7/site-packages (from hidapi->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
mac% which python
/usr/local/bin/python
mac% which pip
/usr/local/bin/pip
mac% which mbed
/usr/local/bin/mbed
mac% python --version
Python 2.7.15
mac% pip --version
pip 9.0.3 from /usr/local/lib/python2.7/site-packages (python 2.7)
mac%

ansondtx20, 2018-05-31:
Also... Feature request: It would be great if we could have an option in mbed-cli that would enable me to see the full paths that mbed-cli is using in my install for python and pip... which binaries are being referenced?  Thanks!

theotherjimmy, 2018-09-27:
@DougAnsonAtARM I think your issue is that `mbed` and `pip` use different installed pythons. As there is no way for mbed CLI to resolve this issue, I'm going to close this issue. Please re-open if you verify that `mbed` and `pip` affect the same python environment and the problem persists.

closed 2018-09-27 by theotherjimmy

zaphodikus, 2019-01-19:
Same. I had things running in 3.7, but, The mbed wrapper invoked python27 for some reason (you see this when you run with a -vv option), then I realised due to installing python 3.7 for a unrelated task after mbed, it caused this. Install mbed again (into the 3.7 environment) and you should be all running again in minutes on the latest and best. Easy resolution.