ARMmbed/mbed-cli #685: custom_targets.json overwrites are not taken into account

drahnr, 2018-06-01:


theotherjimmy, 2018-06-01:
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.7/targets/targets.json#L3454-L3499

Is the offending target definition. It looks like you should be able to remove S132 from the macros, and if SOFTDEVICE_PRESENT is correctly removed, then that should be too.

drahnr, 2018-06-01:
But that is exactly the point - S132 is not removed when done within a `custom_target.json` target definition. It works fine within an `mbed_app.json`

theotherjimmy, 2018-06-01:
@drahnr Does it only affect `S132`? is there any other macro that behaves similarly? If there is we might have a pattern here. Further, any fixes for this behavior (supposing that there is a bug) will come to 5.9.0 or later.

closed 2019-10-31 by ciarmcom