ARMmbed/mbed-cli #688: [mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'

MarceloSalazar, 2018-06-04:
New error is shown when attempting to program the platform with a terminal application already connected: "[mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'"

Mbed CLI 1.7.2

Steps to reproduce:
```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed compile -t GCC_ARM -m K64F -f

...
Total Static RAM memory (data + bss): 12352 bytes
Total Flash memory (text + data): 51513 bytes

Image: ./BUILD/K64F/GCC_ARM/mbed-os-example-blinky.bin
[mbed] Detected "K64F" connected to "/Volumes/DAPLINK" and using com port "/dev/tty.usbmodem1412"
[mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'
```

theotherjimmy, 2018-06-04:
@MarceloSalazar Yep bug.

screamerbg, 2018-06-08:
@theotherjimmy Could you provide more information where this bug is coming from? I'm unable to reproduce it.

theotherjimmy, 2018-06-08:
Sure. Just a sec.

theotherjimmy, 2018-06-08:
here: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L38
and: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L46

NiallBegley, 2018-06-14:
I'm having this same issue, is there currently any work arounds?

MarceloSalazar, 2018-06-15:
Afaik this is not a fatal error.
In my environment, only happens when a serial terminal is open and I attempt to program the device (mbed compile ... -f)

nicbkw, 2018-08-14:
using xenial in chroot on Chromebook
mbed version 1.7.5
mbed sterm
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "DISCO_L072CZ_LRWAN1"
[mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'

fasthazard, 2018-08-24:
@NiallBegley, @nicbkw, and @MarceloSalazar, the workaround for me was to update pyserial. 
My system already had pyserial-2.7 and pip install --upgrade pyserial resulted in pyserial-3.4

Added traceback.print_exc to identify failing calls to self.serial.reset_input_buffer(), etc.

To me it looks like there are two bugs here:
1) switch to MbedTerminal class in #664 looks like it kept some error handling from the older mbed_cdc subroutine, including returning boolean error flag instead of None.
2) installer may not properly identify dependency on pyserial > 3.0 (or even higher)
2a) might be possible to detect version via serial.VERSION and adapt calls accordingly. Or just use the older calls (e.g., flushInput() instead of reset_input_buffer()). At least with pyserial-3.4, it looks like the old calls have been retained.

@screamerbg , what do you think?

Note I already had python installed on my machine via python(x,y) superpackage, and didn't want to break it by installing another python environment. So it's possible that the bug doesn't show up when the default mbed-cli installer options are used...

nicbkw, 2018-09-12:
...updated but still problem - now
using xenial in chroot on Chromebook
mbed version 1.8.0
pyserial 3.4
mbed sterm
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "DISCO_L072CZ_LRWAN1"
[mbed] ERROR: Unknown Error: init() should return None, not 'bool'

closed 2018-10-02 by theotherjimmy

JojoS62, 2018-10-17:
I'm still getting this error when I run 'mbed sterm' or 'mbed compile -f' Ubuntu 18.04:
```
jojo@ubuntu:~/projects/mbed-blinky/mbed-blinky$ mbed sterm
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "DISCO_F469NI"
[mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'
---
jojo@ubuntu:~/projects/mbed-blinky/mbed-blinky$ mbed --version
1.8.2
```
mbed-os is >= 5.10.1.

On Windows 10, it is working and I get no error.

nicbkw, 2018-10-17:
Could be as simple as not having the right permissions to use the serial port, e.g. sudo chmod 666 /dev/ttyACM0

JojoS62, 2018-10-17:
yes, so printing the exception points directly to the problem. Another reason for failure would be an already opened port, this is also caught by the exception and printed as error message.

PR #765 (2018-10-02): Return None from MbedTerminal Constructor
Since we examine `self.serial` to determine the success of the 
constructor, I made all `return False` statements within the constructor
set `self.serial = None`. That way the current code should catch the
failures without any additional changes

Fixes #688
commit 12ee3ce6 (2018-10-02): Return None from MbedTerminal Constructor

Since we examine `self.serial` to determine the success of the
constructor, I made all `return False` statements within the constructor
set `self.serial = None`. That way the current code should catch the
failures without any additional changes

Fixes #688