ARMmbed/mbed-cli #692: Export make_gcc_arm error with v1.7.2 + Python 3

imsherlock, 2018-06-06:
Using `mbed export -i make_gcc_arm` with Python 3 for the K82F target results in the following Runtime Error:

C02T2C9YGTFM:mbed-os-example-blinky ryashe01$ mbed export -i GCC_ARM
Scan: .
Scan: FEATURE_UVISOR
Traceback (most recent call last):
  File "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/project.py", line 278, in <module>
    main()
  File "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/project.py", line 273, in main
    export_path=options.build_dir)
  File "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/project.py", line 100, in export
    app_config=app_config)
  File "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 283, in export_project
    temp = copy.deepcopy(res)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 304, in _reconstruct
    for key, value in dictiter:
  File "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 100, in items
    for k in self.lazy.keys():
RuntimeError: dictionary changed size during iteration
[mbed] ERROR: "/usr/local/opt/python/bin/python3.6" returned error code 1.
[mbed] ERROR: Command "/usr/local/opt/python/bin/python3.6 -u /Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky/mbed-os/tools/project.py -i gcc_arm -m K82F --source ." in "/Users/ryashe01/Development/oob-mbed-os/blinky/mbed-os-example-blinky"

theotherjimmy, 2018-06-06:
https://github.com/ARMmbed/mbed-os/pull/7092 Should have fixed that error. Are you using mbed-os 5.9.0-rc2?

MarceloSalazar, 2018-06-08:
Duplicate of https://github.com/ARMmbed/mbed-cli/issues/694 ?

cmonr, 2018-06-08:
@MarceloSalazar Not quite.

cmonr, 2018-06-11:
@imsherlock Could you share the tool versions that you're using? The instance of Mbed OS inside of the blinky example needs to be v5.9+.

Otherwise, I'm not able to reproduce this on my end using the latest releases of mbed-cli (1.7.3), Python 3 (3.6.5), and Mbed OS checked out to master (Windows has a fix incoming for 5.9.1, but Linux should be fine).

MarceloSalazar, 2018-06-12:
@imsherlock you can just type `mbed ls` inside the blinky project and share the output?

You can use `mbed update mbed-os-5.9.0-oob` to switch to the latest branch and confirm it works for you.

MarceloSalazar, 2018-09-06:
We can't reproduce this issue. Closing.
Please try with 5.10 if you can and reopen if you think it's still a problem. Thanks!

closed 2018-09-06 by MarceloSalazar