ARMmbed/mbed-cli #693: Problem compiling mbed 2.0 programs for NRF51_DK_OTA or NRF51_DK_BOOT

mauriziobanfi, 2018-06-07:
Hello, 
I just set up CLI to compile mbed 2.0 programs. Everything works correctly when I compile for the Nordic NRF51_DK platform. As I attempt to compile for NRF51_DK_OTA (FW not including SoftDevice) or for NRF51_DK_BOOT (FW including the bootloader), the compiler returns the following error message: 

`C:\Users\Maurizio Banfi\mbed_blinky>mbed compile -t GCC_ARM -m NRF51_DK_OTA
Building project mbed_blinky (NRF51_DK_OTA, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [100.0%]: main.cpp
[Fatal Error] platform.h@29,20: device.h: No such file or directory
[ERROR] In file included from ./mbed/aa5281ff4a02/mbed.h:54:0,
                 from .\main.cpp:1:
./mbed/aa5281ff4a02/platform/platform.h:29:20: fatal error: device.h: No such fi
le or directory
 #include "device.h"
                    ^
compilation terminated.

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\Maurizio Banfi\mbed_bl
inky\.temp\tools\make.py -t GCC_ARM -m NRF51_DK_OTA --source . --build .\BUILD\N
RF51_DK_OTA\GCC_ARM" in "C:\Users\Maurizio Banfi\mbed_blinky"
---`

I'm using mbed version 1.7.3. 
Thank you for any suggestion to make it work!

theotherjimmy, 2018-06-07:
Here's my process for examining this:
1) Start at the target definition: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2583-L2588
2) Go up to the inheritance chain until I find the target that's released for Mbed 2: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2388-L2394, https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2375-L2380, https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2259-L2306

None of those devices are released for mbed 2.0.

theotherjimmy, 2018-06-07:
You may want to use Mbed 5 for this project.

mauriziobanfi, 2018-06-07:
The reason why you write "None of those devices are released for mbed 2.0." is because none of my targets and none of their "parents" display the line: "release_versions": ["2", "5"], or "release_versions": ["2", "5"]?

theotherjimmy, 2018-06-07:
@mauriziobanfi That's correct.

mauriziobanfi, 2018-06-07:
@theotherjimmy Thank you! I actually just realized that I obtain the same error even when I attempt to compile for the same targets with the online compiler. With some previous versions of Mbed-2, however, that used to be possible... I will begin studying Mbed-5 :-)

closed 2018-06-07 by mauriziobanfi