ARMmbed/mbed-cli #694: Export make_gcc_arm results in TypeError with v1.7.3 + Python3

atomichan, 2018-06-08:
In producing make file by mbed export command, it fails as follows:

```
C:\work\mbed-os-example-blinky>mbed export -m nucleo_l476rg -i make_gcc_arm
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions,
Python errors will occur when compiling, testing and exporting
---
Scan: .
Traceback (most recent call last):
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\project.py", line 286, in <module>
    main()
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\project.py", line 281, in main
    ignore=options.ignore)
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\project.py", line 102, in export
    app_config=app_config, ignore=ignore)
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\export\__init__.py", line 304, in export_project
    macros=macros)
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\export\__init__.py", line 161, in generate_project_files
    exporter.generate()
  File "C:\work\mbed-os-example-blinky\mbed-os\tools\export\makefile\__init__.py", line 130, in generate
    ctx['asm_flags'][index] = "-I" + ctx['vpath'][0] + "/" + ctx['asm_flags'][index][2:]
TypeError: 'map' object is not subscriptable
[mbed] ERROR: "c:\users\atstom01\appdata\local\programs\python\python36-32\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\users\atstom01\appdata\local\programs\python\python36-32\python.exe -u C:\work\mbed-os-example-blinky\mbed-os\tools\project.py -i make_gcc_arm -m nucleo_l476rg --source ." in "C:\work\mbed-os-example-blinky"
---
```

My environment:
```
C:\work\mbed-os-example-blinky>mbed --version
1.7.3
C:\work\mbed-os-example-blinky>python --version
Python 3.6.5
```

cc @MarceloSalazar @ashok-rao

MarceloSalazar, 2018-06-08:
@atomichan thanks for the report
Can you confirm you switched to the mbed-os-5.9.0-oob branch?
Can you share the output of 'mbed ls'?

MarceloSalazar, 2018-06-08:
I've tried with the 5.9 OOB branch and I'm able to reproduce this issue.
The nucleo_l476rg is marked as supported with make_gcc_arm exporter.

FYI @adbridge @theotherjimmy

atomichan, 2018-06-08:
@MarceloSalazar thank you for the information of reproduction.

Here’s the output of ‘mbed ls’

```
mbed-os-example-blinky (#f38e75b984e3, tag: mbed-os-5.9.0-rc2)
`- mbed-os (#4aa2bf6aec70, tag: mbed-os-5.9.0-rc2)
```

According to the error message, can this issue relate to the change of behavior that map object in python 3.x returns iterator object, not a list?

https://docs.python.org/3.6/library/functions.html#map

cmonr, 2018-06-08:
Will look into it. Should have been fixed with https://github.com/ARMmbed/mbed-os/pull/7092

cmonr, 2018-06-08:
@MarceloSalazar @atomichan Could y'all retry the command with Mbed OS checked out to master?

https://github.com/ARMmbed/mbed-os/pull/7111 is already in the repo and slated for 5.9.1.

atomichan, 2018-06-08:
I retried the command with the master branch of Mbed OS and made sure that it was successful.

cmonr, 2018-06-08:
@atomichan Thanks for the quick check. Going to close since it sounds like #7111 resolves the issue.

cmonr, 2018-06-08:
Actually, going to reopen. Need to verify if #7111 fixes the problem, or #7092. If it's the latter, then somehow it didn't make it into RC2 properly.

cmonr, 2018-06-11:
Confirmed that https://github.com/ARMmbed/mbed-os/pull/7111 solves issue.

closed 2018-06-11 by cmonr