ARMmbed/mbed-cli #695: mbed-cli should report actual failure reason as part of ERROR message.

teetak01, 2018-06-08:
Sometimes it is a bit difficult to identify why mbed-cli failed some operation, especially if it is reported by third-party only showing normal logs.

For. ex.

[mbed] Adding library "mbed-cloud-client" from "https://...."
[mbed] ERROR: Unable to clone repository (....)

Here it would be more useful to give also some error code for ex.

[mbed] ERROR (Forbidden: code 403): Unable to clone repository (...)

I this case the error was caused by too old Git-client in windows (which was still trying to use deprecated weak encryption).

I presume the communication within mbed-cli is relative REST-like, so there should be some error-code to pass forward to customer.

theotherjimmy, 2018-06-08:
> I presume the communication within mbed-cli is relative REST-like, so there should be some error-code to pass forward to customer.

I'm not sure how that plays with `--protocol ssh`. There's probably an error code in either case.

screamerbg, 2018-06-08:
@teetak01 Could you attach the original error you received?

teetak01, 2018-06-08:
mbed-cli really did not output anything more than that and that is from customer reported issue (thus hiding some details).

This is more of a general request to relay some error codes. Especially with quite common issues like missing RSA keys, failing authentication etc. which can happen when working with private repositories.

This might be reproducible if you give invalid rsa keys and try to deploy.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-10-31 by ciarmcom