ARMmbed/mbed-cli #697: Multiple linker scripts detected; Library name xxx is not unique

mlubinsky, 2018-06-08:
**Issue:**  I was able to build this project in the past and I do not see why it fails now:
  mbed compile -t GCC_ARM -m K64F
Building project mbed-os-example-tls (K64F, GCC_ARM)
Scan: .
**Warning: Multiple linker scripts detected**: ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld -> ./tls-client/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld
Scan: mbed
Scan: env
**[ERROR] Library name 'drivers' is not unique** (defined in '/Users/miclub01/GIT/mbed-os-example-tls/tls-client/mbed-os/drivers/mbed_lib.json' and '/Users/miclub01/GIT/mbed-os-example-tls/mbed-os/drivers/mbed_lib.json')
[mbed] ERROR: "/usr/local/opt/python@2/bin/python2.7" returned error code 1.
[mbed] ERROR: Command "/usr/local/opt/python@2/bin/python2.7 -u /Users/miclub01/GIT/mbed-os-example-tls/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM" in "/Users/miclub01/GIT/mbed-os-example-tls"
**Environment:**
  mbed --version       => 1.7.3
 python -V    =>     Python 2.7.15
 MacOS High Sierra 10.13.4

theotherjimmy, 2018-06-08:
You may want to check that Mbed-OS is clean. It seems like you may have pre-K64F-migration stuff mixed with post-K64F-migration stuff.

theotherjimmy, 2018-06-08:
AH! you have 2 mbed-os's one in the project root as `mbed-os` and one in `tls-client/mbed-os`. This generally won't work. You probably don't need the one in tls-client.

mlubinsky, 2018-06-08:
I think it is is because  there are 2 "targets" folders:
 ./mbed-os/targets/
./tls-client/mbed-os/targets/
When I step into ./tls-client/  then the compilation works.

closed 2018-06-08 by mlubinsky