ARMmbed/mbed-cli #698: Mbed tools are not present when creating a new Mbed OS 2 program with --mbedlib

yennster, 2018-06-12:
**mbed-cli version:** 1.7.3

I follow the instructions in the Mbed CLI README documentation for creating a new Mbed OS 2 program (https://github.com/ARMmbed/mbed-cli#creating-a-new-program-for-mbed-os-2) and I receive the following output:

```
~/Repos $ mbed new mbed-classic-program --mbedlib
[mbed] Creating new program "mbed-classic-program" (git)
[mbed] Adding library "builds" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Downloading library build "5aab5a7997ee" (might take a while)
[mbed] Unpacking library build "5aab5a7997ee" in "/Users/jenplu01/Repos/mbed-classic-program/builds"
[mbed] Unpacking library build "5aab5a7997ee" in "/Users/jenplu01/Repos/mbed-classic-program/builds"
[mbed] Updating reference "builds" -> "https://mbed.org/users/mbed_official/code/mbed/builds/5aab5a7997ee"
[mbed] WARNING: Cannot find the mbed tools directory in "/Users/jenplu01/Repos/mbed-classic-program"
---
```

Unlike in the README, the Mbed 2.0 build tools are not downloaded. So, when I try to use commands like `mbed export`, `mbed compile` etc. I get the following output:

```

Tony-YI, 2018-09-08:
same for 1.7.2

theotherjimmy, 2018-09-10:
Thanks for verifying that this bug is still present.

JojoS62, 2018-11-13:
Is this problem still existent? I'v tried it now with actual mbed-cli 1.8.3 and the mbed libname is correct and also the tools are downloaded.
But I prefer to create an empy project and use 'mbed add https://os.mbed.com/users/mbed_official/code/mbed-dev/' for the source version which is much leaner than the mbedlib version.
There I have no tools and 'mbed deploy' does not load them. Mbed deploy detects a mbed classic by checking if file mbed.bld exists. Creating an empty mbed.bld is tweaking the deploy command and it will download the tools. But it would be cleaner if this could be done by a '--force-mbed2-tools' or similar switch. Would this make sense or is there already another trick to achive this?

PhanLeSon03, 2019-02-21:
I am having the same errors, does someone know the solution?

yennster, 2019-02-22:
@PhanLeSon03 Make sure your Mbed CLI is updated: `pip install -U mbed-cli`

If you encounter the `fatal error: mbed.h: No such file or directory` follow the instructions on this thread: https://github.com/ARMmbed/mbed-cli/issues/805