ARMmbed/mbed-cli #700: separate build subdir for each build profile

bmcdonnell-ionx, 2018-06-12:
Feature/behavior request: make the default build dir for `mbed compile` a subdir under `BUILD/`. Name the subdir the same as the name of the build profile file, without extension.

e.g.

| Command                                                       | Build Dir                     |
| ------------------------------------------------------------- | ----------------------------- |
| `mbed compile`                                                | `BUILD/develop/`              |
| `mbed compile --profile debug`                                | `BUILD/debug/`                |
| `mbed compile --profile=myprofiles/superoptimized-cpp11.json` | `BUILD/superoptimized-cpp11/` |

This way users can switch between build profiles without having to rebuild everything, and without having to specify custom build dirs.

(Re: the first example, `mbed-os/tools/profiles/develop.json` is the default build profile, right?)

bmcdonnell-ionx, 2018-06-12:
I suggest parsing from the build profile filename once it is found on the file system, not from the command line. That way, on a case-insensitive filesystem, e.g. `mbed compile --profile debug` and `mbed compile --profile DEBUG` don't create duplicates.

0xc0170, 2018-06-14:
I think this is similar to the one #615 (might be duplicate)

bmcdonnell-ionx, 2018-06-20:
Closed as dupe of #615.

closed 2018-06-20 by bmcdonnell-ionx