ARMmbed/mbed-cli #701: MAX32620HSP OS5 support

amitchone, 2018-06-13:
Hi, 

Trying to compile a project (using the CLI tools) that is verified working with the online compiler using the MAX32620HSP board. 

`mbed compile -m MAX32620HSP -t ARM`
`>> Building project KYM_BIO-HSP (MAX32620HSP, ARM)`
`>> Scan: .`
`>> Scan: mbed`
`>> Scan: env`
`>> Scan: FEATURE_BLE`
`>> Could not compile for MAX32620HSP: Target does not support mbed OS 5`

The board _is_ OS5 compatible. Not entirely sure what I'm doing wrong if there's no issue. Let me know what info would help!

Adam

theotherjimmy, 2018-06-13:
Sorry, That board is not OS5 compatible: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2858

amitchone, 2018-06-13:
That leaves me ever more confused.. The [platform listing](https://os.mbed.com/platforms/MAX32620HSP/) states OS5 compatibility and compilation via the online tool works just fine using OS5?

theotherjimmy, 2018-06-13:
The second bit is a bug. It should fail in the online compiler with the same message as the offline compiler.

amitchone, 2018-06-13:
Is there a published timeline for official inclusion of OS5 support available? Thanks for your help.

theotherjimmy, 2018-06-13:
@amitchone I don't think there are any open PRs that add "5" into the release versions list I linked above.

theotherjimmy, 2018-06-13:
@screamerbg @maclobdell @bentcooke This is venturing into your department.

screamerbg, 2018-06-18:
Bump @maclobdell @bentcooke

amitchone, 2018-06-27:
Regarding the above reference (#7345), could that be due to the removal of support for this platform?

closed 2018-07-06 by amitchone