ARMmbed/mbed-cli #702: [mbed] ERROR: Unable to reset the target board connected to your system.

mtfurlan, 2018-06-18:
I'm getting
```
$ mbed compile --flash
...
[mbed] Detected "NRF52840_DK" connected to "/media/mark/MBED" and using com port "/dev/ttyACM0"
[mbed] ERROR: Unable to reset the target board connected to your system.
[mbed] ERROR: This might be caused by an old interface firmware.
[mbed] ERROR: Please check the board page for new firmware.
```
with return code 1, because `mbed_sterm`  has no way to return true if `sterm` is false.
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L1785

I think just setting `result` to true if `sterm` is false is fine? I did it locally and it fixed my issue, but I didn't check all the places it's called.

theotherjimmy, 2018-06-18:
I'm thinking it should be a bit more nuanced:
```python
    try:
        from .mbed_terminal import MbedTerminal
    except (IOError, ImportError, OSError):
        error("The serial terminal functionality requires that the 'mbed-terminal' python module is installed.\nYou can install mbed-terminal by running 'pip install mbed-terminal'.", 1)

    result = True
    mbed_serial = MbedTerminal(port, baudrate=baudrate, echo=echo)
    if mbed_serial.serial:
        if reset:
            result = result and mbed_serial.reset()

        if sterm:
            # Some boards will reset the COM port after SendBreak, e.g. STLink based
            if not mbed_serial.serial.is_open:
                mbed_serial = MbedTerminal(port, baudrate=baudrate, echo=echo)

            try:
                result = result and mbed_serial.terminal()
            except:
                pass
return result

theotherjimmy, 2018-06-18:
That would require all operations specified to pass for a `True` return value.

mtfurlan, 2018-06-18:
That would be wrong if `mbed_serial.serial` is false.
Also the `MbedTerminal` init tries to return false in a few places as noticed in #688, so some way of handling those errors is probably important, though that may be handled by checking `mbed_serial.serial`, but it doesn't help return a useful error message.

theotherjimmy, 2018-06-18:
@mtfurlan Yep. that's correct. Edited.

mtfurlan, 2018-06-18:
So trying that, I've learned that on my computer, the reset function does not complete successfully. `mbed_sterm` does appear to correctly handle failures now.

In the reset function, `sendBreak` throws a broken pipe error.
Also the catch for that tries to run setBreak, which got removed from the serial library.
http://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.setBreak

As I still don't have working serial output from my dev board, I can't confirm that ignoring a broken pipe is fine, but the internet seems to suggest that it should be fine as it just means that there is nothing reading the other end.

mtfurlan, 2018-06-18:
Ignoring it works fine for me.
Not calling reset at all does fail to reset it (pressing the reset button works fine), so I think that means that calling `sendBreak` and ignoring the broken pipe resets it fine.

screamerbg, 2018-06-18:
Note that the mbed_terminal reset method is (almost) a copy of the htrun reset module. 

E.g. https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L145
```
    def reset(self):
        try:
            self.serial.sendBreak()
        except:
            try:
                self.serial.setBreak(False) # For Linux the following setBreak() is needed to release the reset signal on the target mcu.
            except:
                return False
        return True
```

https://github.com/ARMmbed/htrun/blob/266d18dbccd1724e02e7e7be4bf2133f71cda953/mbed_host_tests/host_tests_plugins/module_reset_mbed.py#L67
```
    def _safe_sendBreak_v2_7(self, serial):
        """! pyserial 2.7 API implementation of sendBreak/setBreak
        @details
        Below API is deprecated for pyserial 3.x versions!
        http://pyserial.readthedocs.org/en/latest/pyserial_api.html#serial.Serial.sendBreak
        http://pyserial.readthedocs.org/en/latest/pyserial_api.html#serial.Serial.setBreak
        """
        result = True
        try:
            serial.sendBreak()
        except:
            # In Linux a termios.error is raised in sendBreak and in setBreak.
            # The following setBreak() is needed to release the reset signal on the target mcu.
            try:
                serial.setBreak(False)
            except:
                result = False
return result
```

I'm wondering if this is a delay issue. E.g. there should be a bit of delay after reset to ensure that the target can recover before the next terminal action.

@mtfurlan Could you please share your operating system and interface firmware version that you run on the nRF52840 DK?

mtfurlan, 2018-06-18:
Debian 9, Linux boethiah 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
I'm running mbed_cli with Python 2.7.13 because that's the default for my system. I could try python 3 if needed.

I'm working with the Nordic-nRF52840-DK, and I just put the mbed firmware 0221 on it this morning.

I updated mbed-cli with pip this morning to 1.7.3, and `mbed-os/mbed.h` says `#define MBED_LIBRARY_VERSION 162`

screamerbg, 2018-06-18:
@mtfurlan Thanks for the prompt response. Python 2.7 is absolutely fine. I'm trying to understand what causes this weird behavior. I'm using Mac OSX and the behavior of Ubuntu Linux VM and the USB stack is very different than the native one. Could you perhaps add a 4 second sleep at the very top of reset()?

mtfurlan, 2018-06-18:
Putting a sleep at the top of reset didn't change anything, I still got a
broken pipe.
I changed `self.serial.setBreak(False)` to
`self.serial.break_condition=False` as per the documentation, to prevent
the `'Serial' object has no attribute 'setBreak'` issue, but that just
throws a second broken pipe error.
I can try setting up a clean OS install on a different device and see if it
does the same thing tomorrow.

screamerbg, 2018-06-22:
@mtfurlan I'm trying to understand where the problem comes from. I was wondering if this might be related to the issue - https://github.com/ARMmbed/mbed-cli/pull/704. Note that this will be available in the Mbed CLI 1.7.4 release (hopefully out today)

mtfurlan, 2018-06-22:
So I just tried the code in the master branch, and I got a lot of
```
[mbed] ERROR: Unknown Error: __init__() should return None, not 'bool'
```
I didn't see anything in the diff from 1.7.3 to master to explain how it was opening the serial port differently, but I didn't look too closely and just applied only the patch #704
That made it go back to
```
[mbed] ERROR: Unable to reset the target board connected to your system.
[mbed] ERROR: This might be caused by an old interface firmware.
[mbed] ERROR: Please check the board page for new firmware.
```

The issue I'm having is that
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L147
is throwing an object of type `termios.error` which is `(32, 'Broken pipe')`
I think ignoring the broken pipe is fine, as the device is probably resetting and works fine in spite of this.

There is a second issue, that doesn't really affect me, but exists is that
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L150
throws `'Serial' object has no attribute 'setBreak'`
This is because it is deprecated: http://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.setBreak

This is the diff from 1.7.3 that I'm using, I just ignored the setBreak issue because I've never seen it.
http://ix.io/1eoq
It includes the patch #704, and a not semantically correct fix for issue #688 because I see that sometimes, mostly when I program twice in a row with not much of a delay. I just don't know which exception to use so I used Exception. Which the tutorial I was reading said not to do.

Sometimes 
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed_terminal.py#L46
happens, and then I need to reset the board manually, but it doesn't happen often, it mostly happens when I flash the board twice in quick succession.

I did get to trying it on a different computer, a ubuntu (16.04.4 LTS (Xenial Xerus)) that is not a fresh install. 
It throws the same broken pipe error from `sendBreak()`. No difference as far as I can see.