ARMmbed/mbed-cli #705: mbed import/add over ssh forces to add a username

geoffrey-vl, 2018-06-21:
As a result of following issue/patch #278 mbed-cli now **always** looks for a username in the url, and if none found it appends the **git** user. At least that's how I understand it.
However this removes the option where username matching can be done through the SSH layer, either by finding a matching username on the remote server or through the SSH config file.

Our internal git server relies on this principle, which (I guess) is a very common setup, because it allows to restrict users to read/write on only the repos that they're allowed to work on. For example external consultants could also have access to this git server. 

One bad way to solve this is by appending your git server's username to the lib file, but that would only work if your not part of a team working on the same project. Unless we overhaul the way users are configured on our git server so that we create a git user per team. The latter would be our administrator's nightmare.

Would it be possible to optionally allow to append a username to the *import* and *add* commands so that at least no username ends up in the urls?

screamerbg, 2018-06-22:
@geoffrey-vl Definitely. I'll introduce this as a small patch to support non-username mode.

screamerbg, 2018-06-22:
@geoffrey-vl I spoke too soon.

Falling back to `git@` username was introduced to handle the way Github and Gitlab do authentication when converting URL from https to ssh.

For example, if I issue `git clone ssh://host` without username, then Git assumes that it should use my local host username, which in the case of Github and Gitlab will always return auth error, e.g.
```
$ git clone github.com:ARMmbed/mbed-cli
Cloning into 'mbed-cli'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
It's rather confusing that Github doesn't allow developers to use different username than `git@`.

To solve your case, I think that you do the following:
* Make sure that protocol conversion is not set via `mbed config protocol --unset` (local), and then `mbed config protocol -G --unset` (global).
* Use standard Git ssh schema, e.g. `mbed import ssh://host/path/repo`

Let me know if this solves the issue.

geoffrey-vl, 2018-06-22:
So actually people should have ssh config setup correctly when using github/gitlab :-) Https is favored anyway, so I don't think it would be such a big effort for others to setup ssh correctly because it was their decision to opt for the less-favored ssh. So in the end I think we should at least be able to override/disable that behavior.

Aside of that, I've never set protocol in mbed config, I'm appending it do my import/add commands. And for my team that seems to be required since .lib files contain https url which fail because we only have ssh setup on our internal server. We're using mbed 1.5.1 though, could that be an issue?

screamerbg, 2018-06-22:
Mbed CLI 1.5.1 didn't handle well certain cases with private repositories. Could you please update to the latest Mbed CLI (1.7.3)?

geoffrey-vl, 2018-06-25:
Mbed version:
`
C:\Users\me\Downloads>mbed-cli --version
1.7.3
C:\Users\me\Downloads>git --version
git version 2.17.1.windows.2
`

Cloning the application repo:
`
C:\Users\me\Downloads>git clone ssh://someserver/myapp
Cloning into 'myapp'...
remote: Counting objects: 1192, done.
remote: Compressing objects: 100% (869/869), done.
remote: Total 1192 (delta 644), reused 585 (delta 263)R
Receiving objects: 100% (1192/1192), 982.07 KiB | 10.34 MiB/s, done.
Resolving deltas: 100% (644/644), done.
`

Deploying without defining the protocol (fails because it tries to fetch over https):
`
C:\Users\me\Downloads\myapp>mbed deploy
[mbed] Adding library "easy-connect" from "https://someserver/easy-connect" at rev #aa81de2177ba
fatal: unable to access 'https://someserver/easy-connect/': Failed to connect to someserver port 443: Connection refused
error: Could not fetch origin
[mbed] ERROR: Unable to clone repository (https://someserver/easy-connect/#aa81de2177ba27d5da5dabf077583c11ef87cefa)
`

Deploying with defining the protocol (fails because it's using the wrong user):
`
C:\Users\me\Downloads\myapp>mbed deploy --protocol ssh
[mbed] Adding library "easy-connect" from "ssh://git@someserver/easy-connect" at rev #aa81de2177ba
git@someserver's password:
`

Running a `mbed import ssh://someserver/myapp` fails because it's also using the git user.

closed 2019-10-31 by ciarmcom