ARMmbed/mbed-cli #709: Board name for Seeed-Arch-max

masoudr, 2018-06-23:
Hi,
I have a custom STM32 board based on `STM32F407VE`. I searched and found out it is the same chip as Seeed-Arch-Max board [link-here](https://os.mbed.com/platforms/Seeed-Arch-Max/). I tried the online compiler with this board and it worked fine.
Now I'm trying to compile my code with cli. I used the `mbed compile -S` command to get supported boards but I can`t find the right name for it. Is there anyway that I pass my processor manually (for example STM32F407VE) or any alternative?
Thanks.

screamerbg, 2018-06-25:
@masoudr The board is called ARCH_MAX. E.g. `mbed compile -m ARCH_MAX`

masoudr, 2018-06-25:
Thanks.I just used `mbed compile - S` that shows different results.