ARMmbed/mbed-cli #711: Configure debugger with VSCode

masoudr, 2018-06-25:
Hi,
I'm trying to configure using `Openocd 0.10.0` with VSCode from [this refrence](https://os.mbed.com/docs/v5.8/tutorials/visual-studio-code.html) on Windows 10. I can run `Openocd` and `arm-none-eabi-gdb` manually with exact configuration without any problem but on VS Code it seems that gdb debugger doesn't run at all and only the debugger server runs.
Here is `launch.json` that generated with `mbed export`.
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "C++ Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}/BUILD/ARCH_MAX/GCC_ARM/${workspaceRootFolderName}.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": false,
            "debugServerArgs": "-f interface/stlink-v2.cfg -f target/stm32f4x.cfg",
            "serverLaunchTimeout": 20000,
            "filterStderr": true,
            "filterStdout": false,
            "serverStarted": "GDB\\ server\\ started",
            "preLaunchTask": "mbed",
            "setupCommands": [
                { "text": "-target-select remote localhost:3333", "description": "connect to target", "ignoreFailures": false },
                { "text": "-file-exec-and-symbols ${workspaceRoot}/BUILD/ARCH_MAX/GCC_ARM/${workspaceRootFolderName}.elf", "description": "load file", "ignoreFailures": false},
                { "text": "-interpreter-exec console \"monitor endian little\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor reset\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor halt\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor arm semihosting enable\"", "ignoreFailures": false },
                { "text": "-target-download", "description": "flash target", "ignoreFailures": false }
            ],
            "logging": {
                "moduleLoad": true,
                "trace": true,
                "engineLogging": true,
                "programOutput": true,
                "exceptions": true
            },
            "linux": {
                "MIMode": "gdb",
                "MIDebuggerPath": "/usr/bin/arm-none-eabi-gdb",
                "debugServerPath": "pyocd-gdbserver"
            },
            "osx": {
                "MIMode": "gdb",
                "MIDebuggerPath": "/usr/local/bin/arm-none-eabi-gdb",
                "debugServerPath": "pyocd-gdbserver"
            },
            "windows": {
                "preLaunchTask": "mbed",
                "MIMode": "gdb",
                "MIDebuggerPath": "C:\\Program Files (x86)\\GNU Tools ARM Embedded\\7 2017-q4-major\\bin\\arm-none-eabi-gdb.exe",
                "debugServerPath": "C:\\openocd-0.10.0\\openocd-0.10.0\\bin-x64\\openocd.exe",
                "setupCommands": [
                    { "text": "-environment-cd ${workspaceRoot}\\BUILD\\ARCH_MAX\\GCC_ARM" },
                    { "text": "-target-select remote localhost:3333", "description": "connect to target", "ignoreFailures": false },
                    { "text": "-file-exec-and-symbols ${workspaceRootFolderName}.elf", "description": "load file", "ignoreFailures": false},
                    { "text": "-interpreter-exec console \"monitor endian little\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor reset\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor halt\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor arm semihosting enable\"", "ignoreFailures": false },
                    { "text": "-target-download", "description": "flash target", "ignoreFailures": false }
                ]
            }
        }
    ]
}
```
And `tasks.json`
```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "0.1.0",
    "name": "mbed",
    "isShellCommand": true,
    "showOutput": "always",
    "problemMatcher": {
        "owner": "cpp",
        "fileLocation": ["relative", "${workspaceRoot}/mbed-os"],
        "pattern": {
            "regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
            "file": 1,
            "line": 2,
            "column": 3,
            "severity": 4,
            "message": 5
        }
    },
    "args": ["compile", "--profile=debug", "-t", "GCC_ARM", "-m", "ARCH_MAX"],
    "linux": {
        "command": "mbed"
    },
    "osx": {
        "command": "mbed"
    },
    "windows": {
        "command": "mbed"
    }
}
```
And here is the output log on debugger:

> 1: (163) LaunchOptions<LocalLaunchOptions xmlns='http://schemas.microsoft.com/vstudio/MDDDebuggerOptions/2014'
> 1: (173) LaunchOptions  ExePath='C:\Users\Masoud\Desktop\mbed_test\mbed-os-example-blinky\BUILD\ARCH_MAX\GCC_ARM\mbed-os-example-blinky.elf'
> 1: (173) LaunchOptions  WorkingDirectory='C:\Users\Masoud\Desktop\mbed_test\mbed-os-example-blinky'
> 1: (173) LaunchOptions  ExeArguments=''
> 1: (173) LaunchOptions MIMode='gdb'
> 1: (173) LaunchOptions  MIDebuggerPath='C:\Program Files (x86)\GNU Tools ARM Embedded\7 2017-q4-major\bin\arm-none-eabi-gdb.exe'
> 1: (173) LaunchOptions  WaitDynamicLibLoad='false'
> 1: (173) LaunchOptions  DebugServer='C:\openocd-0.10.0\openocd-0.10.0\bin-x64\openocd.exe'
> 1: (173) LaunchOptions  DebugServerArgs='-f interface/stlink-v2.cfg -f target/stm32f4x.cfg'
> 1: (173) LaunchOptions  ServerStarted='GDB\ server\ started'
> 1: (173) LaunchOptions  FilterStderr='true'
> 1: (173) LaunchOptions  ServerLaunchTimeout='20000'
> 1: (173) LaunchOptions>
> 1: (173) LaunchOptions    <SetupCommands>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description=''>-environment-cd C:\Users\Masoud\Desktop\mbed_test\mbed-os-example-blinky\BUILD\ARCH_MAX\GCC_ARM</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description='connect to target'>-target-select remote localhost:3333</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description='load file'>-file-exec-and-symbols C:\Users\Masoud\Desktop\mbed_test\mbed-os-example-blinky/BUILD/ARCH_MAX/GCC_ARM/mbed-os-example-blinky.elf</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor endian little"</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor reset"</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor halt"</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor arm semihosting enable"</Command>
> 1: (173) LaunchOptions        <Command IgnoreFailures='false' Description='flash target'>-target-download</Command>
> 1: (173) LaunchOptions    </SetupCommands>
> 1: (173) LaunchOptions</LocalLaunchOptions>
> 1: (225) Starting: "C:\openocd-0.10.0\openocd-0.10.0\bin-x64\openocd.exe" -f interface/stlink-v2.cfg -f target/stm32f4x.cfg
> 1: (20237) <-logout

theotherjimmy, 2018-06-25:
@janjongboom Could you comment?

janjongboom, 2018-06-27:
@masoudr See https://gist.github.com/janjongboom/51f2edbee8c965741465fa5feefe4cf1 - the `serverStarted` command needs to be different for OpenOCD.

masoudr, 2018-06-27:
@janjongboom Thank you. I just added the server args missing `-c init -c \"reset init\"` and edited `serverStarted` command and short `serverArgs` path worked for me. I think the documentation should be fixed for this. It didn't say anything about `serverStarted` command for `OpenOCD`.

loverdeg-ep, 2018-06-27:
@masoudr  Start an issue in the docs repository. I'd like to develop with VS code but have never felt that the documentation was complete enough to make it worth the effort.

masoudr, 2018-06-27:
@loverdeg-ep I will. But I think I still have the issue, now the command can debug my code and chip is getting debugged but when I run the blinky example I get a weird exception:

> C continue: {"threadId":-2}
> 
> 1: (220259) <--exec-continue
> 
> 1: (220266) ->^running
> 
> 1: (220266) ->*running,thread-id="all"
> 
> 1: (220266) ->(gdb)
> 
> 1: (220266) ->&"\n"
> 
> 1: (220266) ->^done
> 
> 1: (220266) ->(gdb)
> 
> openocd: Info : halted: PC: 0x08008cbc
> 1: (220952) ->=breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x08008cba",func="main()",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8",thread-groups=["i1"],times="2",original-location="main"}
> 
> 1: (220954) ->~"\n"
> 
> 
> 1: (220954) ->~"Breakpoint 1, main () at .\\source\\main.cpp:8\n"
> 
> Breakpoint 1, main () at .\source\main.cpp:8
> 1: (220955) ->~"8\t        led1 = !led1;\n"
> 
> 8	        led1 = !led1;
> 1: (220955) ->*stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x08008cba",func="main",args=[],file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"},thread-id="1",stopped-threads="all"
> 
> 1: (220955) <-1024-stack-list-frames 0 1000
> 
> 1: (220965) ->1024^done,stack=[frame={level="0",addr="0x08008cba",func="main",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"}]
> 
> 1: (220965) ->(gdb)
> 
> 1: (220965) 1024: elapsed time 10
> 
> 1: (220966) ->&"\n"
> 
> 1: (220966) ->^done
> 
> 1: (220966) ->(gdb)
> 
> C threads: null
> 
> C stackTrace: {"threadId":-2,"startFrame":0,"levels":20}
> 
> 1: (220976) <-1025-stack-list-arguments 0 0 0
> 
> 1: (220987) ->1025^done,stack-args=[frame={level="0",args=[]}]
> 
> 1: (220987) ->(gdb)
> 
> 1: (220987) 1025: elapsed time 10
> 
> 1: (220987) ->&"\n"
> 
> 1: (220987) ->^done
> 
> 1: (220987) ->(gdb)
> 
> C scopes: {"frameId":1000}
> 
> 1: (221014) <-1026-stack-list-variables 0
> 
> 1: (221020) ->1026^done,variables=[]
> 
> 1: (221020) ->(gdb)
> 
> 1: (221020) 1026: elapsed time 5
> 
> 1: (221020) ->&"\n"
> 
> 1: (221020) ->^done
> 
> 1: (221020) ->(gdb)
> 
>

masoudr, 2018-06-28:
I opened this issue in the [docs repository](https://github.com/ARMmbed/mbed-os-5-docs/issues/611#issuecomment-401087526).

closed 2018-06-28 by masoudr

loverdeg-ep, 2018-06-29:
@masoudr This repo or the mbed-os repo are probably was the correct location for this particular issue.

My comment was in regard to your earlier statement

>I think the documentation should be fixed for this. It didn't say anything about serverStarted command for OpenOCD.