ARMmbed/mbed-cli #716: [ERROR] cannot use a string pattern on a bytes-like object

ValentinoPeluso, 2018-07-09:
Hi,
when I try to compile I get this error:

[ERROR] cannot use a string pattern on a bytes-like object
[mbed] ERROR: "/usr/local/bin/python3" returned error code 1.

I am using Ubuntu 16.04 with Python3.6.0 and mbed-cli 1.7.3. 

I created a test project with the command:
mbed new test

and then I try to compile with:
mbed compile -m auto -t GCC_ARM

Regards.

screamerbg, 2018-07-09:
Cc @cmonr

ValentinoPeluso, 2018-07-10:
Using python 3.5.2 I got:

Traceback (most recent call last):
  File "/home/peluso/Work/arm/test/mbed-os/tools/make.py", line 41, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/home/peluso/Work/arm/test/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/home/peluso/Work/arm/test/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/home/peluso/Work/arm/test/mbed-os/tools/targets/__init__.py", line 557, in <module>
    update_target_data()
  File "/home/peluso/Work/arm/test/mbed-os/tools/targets/__init__.py", line 546, in update_target_data
    in Target.get_json_target_data().items()
  File "/home/peluso/Work/arm/test/mbed-os/tools/targets/__init__.py", line 70, in wrapper
    CACHES[(func.__name__, args)] = func(*args, **kwargs)
  File "/home/peluso/Work/arm/test/mbed-os/tools/targets/__init__.py", line 142, in get_json_target_data
    Target.__targets_json_location_default)
  File "/home/peluso/Work/arm/test/mbed-os/tools/utils.py", line 351, in json_file_to_dict
    object_pairs_hook=OrderedDict)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'

ValentinoPeluso, 2018-07-10:
@theotherjimmy mbed-os-5.9.2

theotherjimmy, 2018-07-10:
I get a different traceback on that version, corrected by https://github.com/ARMmbed/mbed-os/pull/7466

theotherjimmy, 2018-07-10:
you may want to `git clean -xf` in mbed-os. That will clean up any old byte code that you may have generated with an old version of mbed-os (which had that problem).

ValentinoPeluso, 2018-07-11:
@theotherjimmy I tried the following commands:
cd mbed-os
git clean -xf

but I continue to get the same error message when I try to compile.

theotherjimmy, 2018-07-11:
I cannot reproduce the specific error you encountered.

mecwerks, 2018-07-11:
Having the same issue here. Installed mbed cli though mingw, using python 3.4.5, downloaded the mbed-os-example-blinky project and get the same error as @ValentinoPeluso

ilovealbis, 2018-07-16:
Also having the same issue. I was trying to compile the Blinky example code. Using Windows and installed mbed-cli through pip.

### Version
```
> mbed-cli
usage: mbed [-h] [--version]             ...

Command-line code management tool for ARM mbed OS - http://www.mbed.com
version 1.7.5
```
```
> mbed ls
mbed-os-example-blinky (#1cd80b0e70a9)
mbed-os (#62f8b922b420, tags: latest, mbed-os-5.9.2)
```
```
> python --version
Python 3.6.5
```
### Compiler Path
```
> mbed config --list
[mbed] Global config:
GCC_ARM=C:\Program Files (x86)\GNU Tools Arm Embedded\7 2018-q2-update\bin
```
### Command and Output
```
> mbed compile -m NUCLEO_L476RG -t GCC_ARM -vv
---
[mbed] Working path "C:\...\workspace\mbed-os-example-blinky" (program)
[mbed] Exec "c:\...\appdata\local\programs\python\python36-32\python.exe -m pip list -l" in "C:\...\workspace\mbed-os-example-blinky"
Package          Version
---------------- ---------
appdirs          1.4.3
beautifulsoup4   4.6.0
certifi          2018.4.16
chardet          3.0.4
colorama         0.3.9
enum34           1.1.6
fasteners        0.14.1
future           0.16.0
fuzzywuzzy       0.16.0
idna             2.7
intelhex         2.2.1
intervaltree     2.1.0
Jinja2           2.10
jsonschema       2.6.0
junit-xml        1.8
lockfile         0.12.2
MarkupSafe       1.0
mbed-cli         1.7.5
mbed-greentea    1.4.0
mbed-host-tests  1.4.0
mbed-ls          1.4.6
mock             2.0.0
monotonic        1.5
pbr              4.1.0
pip              10.0.1
prettytable      0.7.2
pyelftools       0.24
pyOCD            0.10.0
pyserial         3.4
pywinusb         0.4.2
PyYAML           3.13
requests         2.19.1
setuptools       39.0.1
six              1.11.0
sortedcontainers 2.0.4
urllib3          1.23
websocket-client 0.48.0
[mbed] Exec "c:\...\appdata\local\programs\python\python36-32\python.exe -u C:\...\workspace\mbed-os-example-blinky\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_L476RG --source . --build .\BUILD\NUCLEO_L476RG\GCC_ARM -v" in "C:\...\workspace\mbed-os-example-blinky"
Traceback (most recent call last):
  File "C:\...\workspace\mbed-os-example-blinky\mbed-os\tools\make.py", line 290, in <module>
    ignore=options.ignore)
  File "C:\...\workspace\mbed-os-example-blinky\mbed-os\tools\build_api.py", line 555, in build_project
    toolchain.version_check()
  File "C:\...\workspace\mbed-os-example-blinky\mbed-os\tools\toolchains\gcc.py", line 118, in version_check
    match = self.GCC_VERSION_RE.search(stdout)
TypeError: cannot use a string pattern on a bytes-like object
[mbed] ERROR: "c:\...\appdata\local\programs\python\python36-32\python.exe" returned error.
       Code: 1
       Path: "C:\...\workspace\mbed-os-example-blinky"
       Command: "c:\...\appdata\local\programs\python\python36-32\python.exe -u C:\...\workspace\mbed-os-example-blinky\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_L476RG --source . --build .\BUILD\NUCLEO_L476RG\GCC_ARM -v"
```

Albis-dev, 2018-07-16:
I looked into this issue and found a temporary fix.

#### Original mbed-os\tools\toolchains\gcc.py
```python
# this function checks the gcc compiler version
 def version_check(self):
        # run "%GCC_ARM_PATH%\arm-none-eabi-gcc --version" and store the output into "stdout"
        stdout, _, retcode = run_cmd([self.cc[0], "--version"], redirect=True)
        msg = None
        match = self.GCC_VERSION_RE.search(stdout) # this line is the problem
        ...
```
#### Output value of "arm-none-eabi-gcc --version"
```
> arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Basically gcc.py uses regex to extract the substring '7.3.1' but run_cmd() returns the value to byte, not string. That's why we get a TypeError. 

#### Quick Fix. Modify gcc.py
```python
    def version_check(self):
        stdout, _, retcode = run_cmd([self.cc[0], "--version"], redirect=True)
        msg = None
        stdout = stdout.decode("utf-8") # ADD THIS LINE
        match = self.GCC_VERSION_RE.search(stdout)
```

theotherjimmy, 2018-07-16:
@Albis-dev Maybe you missed the comment above where I submitted a PR to mbed-os with a fix for the very issue you ran into. https://github.com/ARMmbed/mbed-os/pull/7466 is the PR.

cmonr, 2018-07-16:
@ValentinoPeluso @Albis-dev @mecwerks Since the PR that fixes the issues is now on master (and in the mbed-os-5.9 branch), would y'all mind updating your installations and verifying that the issue was fixed?

theotherjimmy, 2018-07-16:
@ValentinoPeluso @Albis-dev @mecwerks Note that it's on master of mbed-os, and that you need to update mbed-os to get the fix.

Albis-dev, 2018-07-16:
@cmonr Can confirm it's working now.