ARMmbed/mbed-cli #721: sterm crashes

janjongboom, 2018-07-25:
Running something based on Mbed Cloud Client:

```
janjon01:~/repos/connected-lights-cloud (master) $ mbed sterm --baudrate 115200
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "K64F"
--- Terminal on /dev/tty.usbmodem1462 - 115200,8,N,1 ---
[EasyConnect] IPv4 mode
[EasyConnect] Using Ethernet
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@2/2.7.15/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@2/2.7.15/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/usr/local/lib/python2.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
SerialException: read failed: [Errno 6] Device not configured
```

screamerbg, 2018-07-25:
This is because the device disconnected the serial com port while the terminal was still open. The device exception cannot be handled by mbed CLI as it happens in a thread initiated by the pyserial library. In other words, it's a known issue that disconnecting the serial com port (or the device itself) while stern is still open causes python error and at the moment there's no workaround.

screamerbg, 2018-09-10:
@janjongboom I don't experience this issue with my K64F. Are you by chance having another process/program running which has the com port open?

janjongboom, 2018-09-12:
@screamerbg I can't really recall the exact conditions, but I doubt it.