ARMmbed/mbed-cli #725: Should have an easy way to update Mbed OS in exported project

janjongboom, 2018-07-27:
I'm writing an application note for MDK and this is a real issue I encounter.

1. Find a project.
2. Export to MDK.
3. Now I want to update Mbed OS.

Because mbed-os is not under source control the normal `update`/`sync`/`deploy` things don't work. I need to:

1. Remove mbed and mbed-os folders.
2. Remove mbed.lib / mbed.bld / mbed-os.lib.
3. Run `mbed add mbed-os`.

Could we have some command that magically clears up all old Mbed folders and adds the latest?

Alternatively, a version of `mbed deploy` which will also work if the other folders are not under SCM could also work (but would need a big **WARNING** sign somewhere to avoid people losing data).

theotherjimmy, 2018-07-27:
@janjongboom You're specifically talking about exports from the online compiler/os.mbed.com. Exports from Mbed CLI would just have to `mbed update` & `mbed export` again.

janjongboom, 2018-07-30:
@theotherjimmy Yes, exports from Online Compiler. Now I need to have stuff in my documentation like:

```
Exported projects contain only a subset of Mbed OS 5 and cannot be managed by Mbed CLI. To resolve this:
5. Open a command prompt or terminal window, and navigate to the location where you unzipped the folder.
6. Run:
           rmdir mbed mbed-os /S
7. Then run:
mbed deploy
```

theotherjimmy, 2018-07-30:
That is the workaround. If you have mbed-cli, what's the point of doing an export from the online compiler?

janjongboom, 2018-08-01:
@theotherjimmy How else would I move from Online Compiler to MDK? This is not about importing existing examples, it's about migrating a codebase that I started online to MDK.

theotherjimmy, 2018-08-01:
@janjongboom, That's a very good point. We should consider having a way to pull private examples in Mbed CLI, and fix this particular problem.