ARMmbed/mbed-cli #726: Downloading library build, takes very long, progress bar?

janjongboom, 2018-07-27:
On an Mbed 2 project I'm running `mbed deploy`.

Then it says:

```
[mbed] Downloading library build "a330f0fddbec" (might take a while)
```

It's been doing this for over 10 minutes already. I have a very fast internet connection (200 mbps). What is it doing and can I have a progress bar similar to when we clone Mbed OS?

theotherjimmy, 2018-07-27:
@janjongboom Good suggestion. We could have a progress bar to indicate to users "get a cup of coffee".

closed 2019-10-31 by ciarmcom

josuah, 2023-03-08:
This might be caused by this instead:
https://github.com/ARMmbed/mbed-cli/issues/968

We might get a rush of users encountering this bug as soon as Ubuntu switches from python2 to pyhon3 by default.