ARMmbed/mbed-cli #729: #pragma message will never be printed

drahnr, 2018-08-03:
#warning works, but I assume `#pragma message` goes to `stdout` which is consumed by mbed cli

0xc0170, 2018-08-03:
Can you provide examples, what works and what does not?  What toolchains?

`#pragma message string` - this one? it's during compilation (is it only supported in GCC ?) so should not use stdout (its not runtime message).

drahnr, 2018-08-03:
gcc 7.0.0 and gcc 7.1.0

works:
```
#warning foo bar
```

does not:
```
#pragma message("foo bar")
#pragma message(foo bar)
```

theotherjimmy, 2018-08-03:
@drahnr those messages show up with `-v`. without the verbose option, the mbed build tools try to limit the amount of terminal noise you get to just warnings and errors. This is by design.

drahnr, 2018-08-03:
The issue with `-v` is, that one gets flooded with compiler defines / commandine options so the desired effect of a message is either being lost in too much information or not printed at all.
Use case: Easy tracking of certain important configuration options without the noise without generating warnings which would error our build.

theotherjimmy, 2018-08-03:
@drahnr I'm a little shocked that you managed to find a target that does not emit warnings. We could parse the `#pragma message` stuff and emit it in the same way as warnings/errors. That would take a bit of work thought, as it would have to work for all 3 supported compilers.

drahnr, 2018-08-03:
@theotherjimmy it only works with compiling many chunks of the final application as libs, mbed-os itself is for sure not free of warnings...

theotherjimmy, 2018-08-06:
@drahnr I think we could also implement a "filter warnings" feature if that would be useful to you. It would have to be for 5.10.

drahnr, 2018-08-06:
@theotherjimmy can you be a little more specific what exactly would be the your usecase? Right now I am not sure how that helps my case: I have a lib which has to be warning free -> filtering warnings won't change the output, or maybe I just did not get what you meant :]

theotherjimmy, 2018-08-06:
@drahnr We could add a filter that ignores warnings within certain paths. So you could do something like `mbed compile --no-warnings mbed-os` and not get warnings from mbed-os.

drahnr, 2018-08-06:
@theotherjimmy Independent of above's discussion: That 'd be great!

The initial problem of not seeing `#pragma messages` is still there though, filtering warnings would only help if I would make them all warnings (which our compilation would prevent).

theotherjimmy, 2018-08-06:
@drahnr I'll be happy to add something like "[INFO]" for messages, if we have a way to find them in all 3 compilers, and we can document how to use it.

closed 2019-10-31 by ciarmcom