ARMmbed/mbed-cli #733: Mbed-cli does not finish driver installation for nRF52-DK

Navis-Raven, 2018-08-10:
At the end of installation of mbed-cli, it asks me to connect an arm mbed enabled microcontroller, so I do it with my nRF52-DK board. However after clicking OK, it sais that "no mbed microcontroller found"

[image]

Navis-Raven, 2018-08-10:
Note that I've used one hour before SEGGER IDE and it updated automatically the nRF52-DK firmware

cmonr, 2018-08-15:
How was mbed-cli installed in this instance? Was it with our Windows installer? Which version of Windows?

Navis-Raven, 2018-08-15:
Yes it was the windows installer. It asked me to plug the arm mbed enabled microcontroller, and I plugged the nRF52-DK and pressed ok, but it prompted this error message.  

My Windows version is Windows 7. 

at the end, I installed it one hour ago with pip, and it didn't asked me to plug a microcontroller

cmonr, 2018-08-22:
@Navis-Raven That's good to know. I'm going to keep this open for now, since this sounds like an installer issue, and it would be useful to know if this is a Windows 7 only issue otr or not.

theotherjimmy, 2018-08-23:
@Navis-Raven The windows installer is designed to be a "one stop shop" for all things related to Mbed CLI. This includes drivers. The pip module is just the CLI, without any supporting drivers or similar. It's probably possible to catch the crash.

Navis-Raven, 2018-08-27:
I have the same problem when trying to install mbedWinSerial_16466.exe on a Windows 8.1 for my nRF52-DK cardboard

[image]

Navis-Raven, 2018-08-29:
I've just tried with STM32 B-L475E-IOT01A (DISCO-L475VG-IOT01A) board, it does not work as well !!!

So the problem is not a problem of a particular OS (same problem is in windows 7 as well as on windows 8.1), nor a problem of dev board (same problem on both of nRF52-DK and B-L475E-IOT01A), it's a more general problem and critical bug !

So I will change the title in function of these results.

theotherjimmy, 2018-08-29:
@Navis-Raven @cmonr The "mbed serial driver" needs a daplink board, probably. @Navis-Raven You're using boards that don't have daplink.

Navis-Raven, 2018-08-29:
@theotherjimmy Thanks for the remark

For the nRF52-DK:
As explained, here: http://docs.platformio.org/en/latest/platforms/nordicnrf52.html#debugging
And here, http://docs.platformio.org/en/latest/plus/debug-tools/cmsis-dap.html#debugging-tool-cmsis-dap

There is already a CMSIS-DAP on board debugger on the nRF52-DK.

For the DISCO-L475VG-IOT01A I will make another test on my windows 7 computer later. And we will see if this a problem specific to nRF52-DK or not.

theotherjimmy, 2018-08-29:
@Navis-Raven The installer may be looking for a Particular vendor and product ID pair. The NRF52-DK has a Segger J-Link with a different vendor and product ID.

Navis-Raven, 2018-08-29:
No it has a segger J-Link AND a CMSIS-DAP on-board debugger

[image]

bridadan, 2018-08-29:
Hi @Navis-Raven, it's true the debugger on board supports either CMSIS-DAP or J-LINK, though the board usually ships configured to use J-LINK. You can verify this by looking on the storage device that appears on your computer when you plug in the usb cable. If you do not see an `MBED.HTM` file, then it is probably using JLINK.

If you'd like to stick with J-LINK, you can install the J-LINK drivers from Segger's website. The mbed serial driver is specifically for boards using CMSIS-DAP. If you'd like to use this debugger, you can find the necessary firmware and instructions here: https://armmbed.github.io/DAPLink/?board=Nordic-nRF51822

(I know the page says "nrf51822", but it should work fine for the nrf52-DK).

Navis-Raven, 2018-08-29:
As you can see here there is already the MBED.HTM file in my nRF52-DK board and it's declared as "mbed enabled"

[image]

I'm following platformIO installation guide (because I'm using platformIO ecosystem) (https://docs.platformio.org/en/latest/plus/debug-tools/cmsis-dap.html#debugging-tool-cmsis-dap) and it says to install windows serial driver

[image]
And it is supposed to work with 0 configuration as soon as you install the drivers (this is the main goal of PlatformIO). 

> (I know the page says "nrf51822", but it should work fine for the nrf52-DK).
For this you'd better contact the webmaster of https://armmbed.github.io and suggest them this update

Navis-Raven, 2018-08-29:
As for the website you gave me, it's the same problem: it says to install "serial port driver" for Windows 8 or older (And I have Windows 7 and another PC with Windows 8.1, I think the webmaster forgot to add the ".1" since Win 8.1 is the same as Win 8 except the GUI).  

[image]

That's it the same driver who displays the bug. Same protocol same issue. 
[image]

Navis-Raven, 2018-08-29:
@bridadan  As for the website you gave me, it's the same problem: it says to install "serial port driver" for Windows 8 or older (And I have Windows 7 and another PC with Windows 8.1, I think the webmaster forgot to add the ".1" since Win 8.1 is the same as Win 8 except the GUI).  

[image]

That's it the same driver who displays the bug. Same protocol same issue. 
[image]

bridadan, 2018-08-29:
@Navis-Raven I realize I said that if the "Mbed.htm" was on the board's storage, then its using CMSIS-DAP. In this case, it  looks like the J-Link firmware also has a copy of the "Mbed.htm" file (all J-Link firmwares are customized for the specific board, so its hard to say for sure what files will be present).

Really the give away that this board is using JLINK is **the "Segger.html" file present**. That takes you to the JLINK product page. This file is not present when CMSIS-DAP is the configuration in use.

**So**, since you want to install the Mbed Serial Driver, you'll want to follow the instructions under the section "INTERFACE FIRMWARE UPDATES" (above the "FREQUENTLY ASKED QUESTIONS" section you posted) to activate the CMSIS-DAP configuration.

Once you've updated your board, you should then be able to install the driver.

Figuring out these debugger firmware versions can be tricky, but the good news is you only have to do it once. I have completed this process many many times, so please let me know if you have any more problems and I'll do my best to help.

theotherjimmy, 2018-08-29:
@Navis-Raven I said *DAPLink* and you're saying *CMSIS-DAP*. These are different things:
 * CMSIS-DAP is a protocol, as you correctly pointed out
 * DAPLink is an implementation of CMSIS-DAP, and more, on some ARMv6 and ARMv7 MCUs

I'm claiming that the "mbed serial driver" works with DAPLink devices. Your counter point was that  you tried a NRF52-DK, and DISCO-L475VG-IOT01A. Neither of the debuggers on those boards run DAPLink by default. They run J-Link and STLink, respectively. Therefore, the "mbed serial driver" may not install when they are connected. We agree then that J-Link and STLink do not work with the "mbed serial driver" installer.

Furthermore, PlatformIO is not Mbed OS or Mbed CLI. I cannot vouch for the correctness of their documentation, and I can't easily fix it. If you could point to Mbed OS or Mbed CLI resources that conflate DAPLink and CMSIS-DAP, I will happily correct them.

Navis-Raven, 2018-08-29:
@theotherjimmy  I'll answer your comment later.

Only small thing has changed on the installer of mbed-cli: It has been renamed from mbed_installer_v047 to Mbed_installer_v0.4.7 

[image]

But nothing changed, the bugs are still here, except that it introduced a small documentation issue to fix, 

[image]

Let me remind all the last screenshot of the installation process
[image]

[image]

And the failure
[image]

Navis-Raven, 2018-08-29:
Now a recap of the tests done: 

### Mbed-cli installation
nRF52-DK on windows 7: FAIL
nRF52-DK on windows 8.1: FAIL

DISCO-L475VG-IOT01A on windows 7: FAIL
DISCO-L475VG-IOT01A on windows 8.1: FAIL

### serial port driver installation
nRF52-DK on windows 7: FAIL
nRF52-DK on windows 8.1: FAIL

Navis-Raven, 2018-08-29:
Ok, that's it I found from where is the bug. 
As I suspected and **I WAS RIGHT** the bugs of both of mbed-cli and "serial-port-driver" **are stritly the same** !

Look at the image:
[image]

As you see for any cardboard, when you install mbed-cli with the windows installer it installs also the "serial port driver" independantly of the debugger choosen or the dev board choosen, and it fails systematically. 

@bridadan thanks for your answer, I've suspected one time that the installation segger tools has modified the firmware of my nRF52-DK, however after I read this webpage again (https://os.mbed.com/docs/v5.9/tools/installing-with-the-windows-installer.html) I remind that I've tried the installation of Mbed-cli and so "serial port debugger" before using and installing segger studio and tools and it fails. Furthemore, as you can see the installation of "serial port driver" cames automatically with mbed-cli installation and wit any board or debugger. **And normally it should install if the board is an arm mbed enabled one, independatly of the debugger or dev board**. My dev boards are both arm mbed boards. 

> Figuring out these debugger firmware versions can be tricky, but the good news is you only have to do it once. 

@bridadan  My goal is not only to make the PC working, since I can abord using these tools and switch with any other dev board/tools which works with more simplicity. By openning an issue here, my goal is to make arm resolving their bug, and this one is a major and critcal one since we cannot even install this driver. Furthemore, I think about the following user who will have to do some gymnastics with the unclear and bugged "getting started protocol" which is not up-to-date and bugs for most of the tools (except segger studio)

Navis-Raven, 2018-08-29:
This is the proof that one of the first thing I did on my nRF52-DK is trying to install mbed-cli and "serial port debugger" **BEFORE** any possible modification by Segger J-Link tools. SO, this is not linked to segger, this is a bug of **ARM SERIAL PORT DRIVER INSTALLER. PLEASE ARM CORRECT THE BUGS OF YOUR TOOLS. THANK YOU !**
[image]

Navis-Raven, 2018-08-30:
Nota Bene: It is probable that this bug is specific to windows 8.1 and previous versions (windows 7 for instance), and not applicable to windows 7, because as said here: https://armmbed.github.io/DAPLink/?board=Nordic-nRF51822 , it is probable that there is no need for drivers on windows 7. 

[image]

theotherjimmy, 2018-08-30:
@Navis-Raven 

> No, the first one runs J-Link and also CMSIS-DAP, and the second only ST-LINK. 

The first one run J-Link out of the box and can run _DAPLink_ if you flash it. I don't see how that's different from what I said:
>> They run J-Link and STLink, respectively. 

My claim is that the "mbed serial driver" only works with _DAPLink_. Please stop saying that I'm incorrect and sighting evidence that does not contradict that statement.

>This is the proof that one of the first thing I did on my nRF52-DK is trying to install mbed-cli and "serial port debugger" BEFORE any possible modification by Segger J-Link tools.

No. That's proof that you downloaded the Setup_EmbeddedStudio... application before you downloaded the mbed CLI installer.

> ** PLEASE ARM CORRECT THE BUGS OF YOUR TOOLS. THANK YOU !**

If this is all that you were after, we would not have ~20 comments in this thread. You seem to be adamant that we assign blame to Arm. I'm of the opinion that a bug is a bug, no matter who is at fault, and that attempting to assign blame is unproductive and should be avoided as it tends to be taken as a personal attack. Please drop the blame assignment and shouting (caps-lock usage), and help us figure out exactly what's going wrong.

Blame asside, the bug here is probably with Non-DAPLink boards + "mbed serial driver" installer. I would like to verify that by testing the DAPLink board + "mbed serial driver" installer on your machine. This will rule out the possibility that it's something with your setup, which we have no evidence of yet.

Navis-Raven, 2018-08-30:
@theotherjimmy From the beginning you act like the fault was not a bug but a misuse of arm tools by me, whereas I followed strictly the arm mbed process in one hand, and platformIO process 1 or 2 weeks later in the other hand on two PC with two differents setupsin two different places (home and at work) and following two different tutorial, and you are still thinking that the problem is my setup : "This will rule out the possibility that it's something with your setup"

And this "** PLEASE ARM CORRECT THE BUGS OF YOUR TOOLS. THANK YOU !**" is not a blame, but a call. Where are the responsibles of this repository ? Because I don't see them here. Nobody is testing on their hardware and PC, I'm the only one doing this here. 

> No. That's proof that you downloaded the Setup_EmbeddedStudio... application before you downloaded the mbed CLI installer

You can still joke if you want, but I remember to have tried installing mbed-cli on my PC as soon as I downloaded it, and without success. 

> My claim is that the "mbed serial driver" only works with DAPLink.

If it was the case the arm mbed instructions will not invite us to install "mbed serial driver" for any board (it installs as soon as we install mbed-cli), and also, for PlatformIO, Please verify your claims as I did by searching the proofs into the documentation, At the moment the official and officious documentation contradicts your claim. (read my comments). You can use greenshot for that. 

> I would like to verify that by testing the DAPLink board + "mbed serial driver" installer on your machine.

I have no "DAPLink board". I have only two arm boards. but arm employees if they are here should have many boards and PC's with windows 7 and 8.1. 

>  the bug here is probably with Non-DAPLink boards + "mbed serial driver" installer

This is possible, but did you made some tests with your DAP-Link or non DAP-Link boards, and mbed-serial-driver installer and one or two PC with ? please share with us if it is the case. 

My theory is that is a problem with "mbed serial driver" installer for winfows 8.1 and inferior.

theotherjimmy, 2018-08-30:
> From the beginning you act like the fault was not a bug but a misuse of arm tools by me, whereas I followed strictly the arm mbed process in one hand, and platformIO process 1 or 2 weeks later in [...]

Sure. I'm trying to rule out all other possibilities. I'm trying to verify that we have a bug here. We seem to: The installer assumes that you want to install the mbed serial driver. If a misuse is the problem, we should improve the documentation to indicate that users should not do whatever caused the problem, and/or improve the installer so that it is harder to "shoot yourself in the foot". Looking back, my wording did not convey my intentions very well; sorry about that. I don't mean to blame you, instead I mean to figure out where a change needs to be made: documentation, installer code or both.

Let's take this down a notch and try to get to the root of the problem.

---------------------

> If it If it was the case the arm mbed instructions will not invite us to install "mbed serial driver" for any board (it installs as soon as we install mbed-cli),

[official documentation](https://os.mbed.com/docs/v5.9/tutorials/windows-serial-driver.html) in case we're talking about different things. I agree, the documentation does not support my claim. I would like to get to what's actually the case, and amend the documentation if appropriate.

> I have no "DAPLink board".

Thanks! @bridadan @cmonr Could you verify that the mbed serial driver is causing issues by testing with a non-daplink and a daplink board to support/contradict my hypothesis? We can update the documentation to instruct users to "not install the mbed serial driver when they don't have a DAPLink board" if my hypothesis is correct. We could also make the installer smarter with a bit of mbedls firmware version detection as @bridadan suggested to me offline.

> My theory is that is a problem with "mbed serial driver" installer for winfows 8.1 and inferior.

Mine too. I specifically think that despite what the documentation says, the mbed serial driver installer only supports DAPLink boards.

screamerbg, 2018-09-10:
@cmonr @bridadan Any follow up on this?

screamerbg, 2018-09-10:
CC @arekzaluski

Navis-Raven, 2019-10-31:
What the hell, this is a defect and not a feature request. but more than that, it is an ancient and critical bug. 

Due to theses incapacity to solve critical bugs in Arm company, I migrated to other technologies such as platformIO ecosystem to uses Arm Cortex microcontrollers