ARMmbed/mbed-cli #736: An extension added to map file in mbed-cli v1.7.5

artokin, 2018-08-22:
After updating mbed-cli to version 1.7.5 and building a application a map-file with extension `.map` does not exists anymore in the build directory. There are `x.map.csv`, `x.map.html`, `x.map.json` and `x.map.old`.

Some of our test jobs are failing as they expect to have `.map` file available in the build directory.

Is this an error or intended behaviour?

theotherjimmy, 2018-08-22:
@artokin That behavoir is not controlled by Mbed CLI. It's probably related to https://github.com/ARMmbed/mbed-os/pull/7590 as it contains a move of the map file (https://github.com/ARMmbed/mbed-os/pull/7590/files#diff-f9cb941bde5647a5763e18fc220efc51R830)

theotherjimmy, 2018-09-11:
@artokin Could you expand on what you're using the map files for?

artokin, 2018-09-12:
The map files were stored by CI job but now the CI jobs have been updated.

theotherjimmy, 2018-09-12:
I'm sorry that my PR broke your CI. I'm glad you were able to work around it.

theotherjimmy, 2018-09-12:
OH! You could set the environment variable `MBED_COMPARE_FIXED` to `True`. Then it would not move the map file on you. Sorry I did not think of that earlier.

theotherjimmy, 2018-09-24:
To directly answer your question: intended behavior.

closed 2018-09-24 by theotherjimmy