ARMmbed/mbed-cli #738: Import fails with utf8 error

floxch, 2018-08-26:
Hi,
when trying to import the following project via mbed import I get an error:
`mbed import http://os.mbed.com/users/simon/code/USBMIDI_HelloWorld/`

The error is as follows:
`[mbed] ERROR: Unknown Error: 'utf8' codec can't decode byte 0xc4 in position 13: invalid continuation byte`

bridadan, 2018-08-27:
Hi @floxch, I was able to import the project on Windows 10 using the latest Mbed CLI (1.7.5) and Python 2.7.13. Could you provide the following info?

- What version of Mbed CLI are you running?
- What version of Python are you using?
- What OS are you using?

floxch, 2018-08-27:
Hi bridadan, thanks for your reply.

- mbed CLI 1.7.2 (latest version from the installer)
- Python 2.7.14
- Windows 8.1

bridadan, 2018-08-27:
I just tried with 1.7.2, still no problems importing, but I'm not using the installer. I don't have a Windows 8.1 machine to test on unfortunately. If you haven't already done so, it'd be worth trying to reimport it (be sure delete any created directories from the previous run of `mbed import`).

theotherjimmy, 2018-08-28:
@floxch It may be related to the directory you're calling `mbed import` from. Do you have a non-utf-8 system encoding? do you have non-utf8 characters in the path you're calling `mbed import` from?

floxch, 2018-08-28:
I just called it directly from my C-drive - same problem :(

screamerbg, 2018-09-10:
@floxch Could you try importing with `-v` option and paste the output here please?

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

mark-edgeworth, 2019-12-04:
This should be fixed by the next version of mbed-cli that we are preparing for release now. Issue looks to be python-2 reated. I have repeated the bug in a py2 environment (on a mac) but the import succeeds ok on python3.