ARMmbed/mbed-cli #742: mbed test --unittests requires a target and a toolchain to be specified

lorjala, 2018-09-05:
mbed test --unittests fails to run if target and toolchain are not specified or .mbed file with those values do not exist.

Commands like `mbed test --unittests -m K64F -t GCC_ARM` can be confusing for users and in CI since unit tests do not use any target or toolchain (gcc compilers are used).

lorjala, 2018-09-05:
@OPpuolitaival @theotherjimmy Can we do anything about this?

Having different tools under the same command with shared options can cause inconsistencies like this.

OPpuolitaival, 2018-09-05:
@theotherjimmy can we make target and toolchain optional and check in case of icetea or greentea? Then mbed unitttest command looks better.

OPpuolitaival, 2018-09-05:
Fix: https://github.com/ARMmbed/mbed-cli/pull/743

theotherjimmy, 2018-09-05:
Thanks for the bug report. 1.8.0 will release with the fix in #743

closed 2018-09-05 by theotherjimmy