ARMmbed/mbed-cli #745: No ability to select API host with `mbed dm` command

dhwalters423, 2018-09-06:
Currently, the Mbed CLI tool defaults to the production api.mbedcloud.com instance of Pelion Device Management.

While this is okay for most customers, there should be an optional argument `-s` that allows you to specify the instance of Pelion Device Management, such as api-os2.mbedcloudstaging.net.

There are 2 reasons for this:

- Internal testing (not critical)
- Support for Pelion On Premises deployment, which will have a different DNS address

yogpan01, 2018-09-06:
Agreed with @dhwalters423. We have requirement to support Cloud on-prem solution where the Cloud deployment is not Production but customer specific deployment.

theotherjimmy, 2018-09-06:
@dhwalters423 does `mbed config -G CLOUD_API_HOST` not work?

theotherjimmy, 2018-09-06:
@dhwalters423 Why `-s`?

yogpan01, 2018-09-06:
@theotherjimmy `-s` would represents own cloud environment. API key is specific to account inside a given environment.

dhwalters423, 2018-09-06:
@theotherjimmy -G CLOUD_API_HOST is not documented anywhere.

theotherjimmy, 2018-09-06:
@dhwalters423 We can fix the documentation issue.

@yogpan01 I was asking why `-s` was picked over other letters, not what it's supposed to mean. Sorry for the confusion.

dhwalters423, 2018-09-06:
@theotherjimmy sounds good to me.

adbridge, 2018-09-10:
@dhwalters423 Is there a PR addressing this issue ?

theotherjimmy, 2018-09-12:
@cmonr They're not new options. They're just re-used options.

closed 2018-09-14 by cmonr